PX4/PX4-Avoidance #563: Global planner hangs at Planner abort: missing required data

aydal, 2020-03-01:
Running global planner on my system, it randomly hangs and begins showing `Planner abort: missing required data`. Any idea of why it happens?

```
roslaunch global_planner global_planner_stereo.launch
... logging to /home/vizzbee/.ros/log/23e75578-5b85-11ea-9cff-185680ce2426/roslaunch-vizzbee-TUF-GAMING-FX504GD-FX80GD-18769.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://vizzbee-TUF-GAMING-FX504GD-FX80GD:34995/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /global_planner_node/frame_id: local_origin
 * /global_planner_node/pointcloud_topics: ['/stereo/points2']
 * /global_planner_node/robot_radius: 0.5
 * /global_planner_node/start_pos_x: 0.5
 * /global_planner_node/start_pos_y: 0.5
 * /global_planner_node/start_pos_z: 3.5
 * /global_planner_node/world_name: /home/vizzbee/cat...
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://:14540@loca...
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.0003490659 // 0...
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: local_origin
 * /mavros/local_position/tf/child_frame_id: fcu
 * /mavros/local_position/tf/frame_id: local_origin
[... 412 lines cut ...]
[ INFO] [1583043817.121229421, 14.000000000]:   Received empty path

[ INFO] [1583043817.121386328, 14.000000000]:   Received empty path

[ INFO] [1583043817.164452530, 14.040000000]: ========== Set goal : (-1067129,-5227321,0) ==========
[ INFO] [1583043818.144801526, 15.000000000]: Goal position is occupied
[ INFO] [1583043818.144953562, 15.000000000]: Failed to find a path
Total time: 0.71 ms 
[ INFO] [1583043818.145178434, 15.000000000]:   Received empty path

[ INFO] [1583043818.145350657, 15.000000000]:   Received empty path

[ INFO] [1583043818.146357243, 15.000000000]: WP: item #0* F:3 C: 22 p: 15.000000 0.000000 0.000000 nan x: 47.397709 y: 8.546704 z: 5.000000
[ INFO] [1583043818.147657862, 15.004000000]: WP: item #1  F:3 C: 16 p: 0.000000 0.000000 0.000000 nan x: 47.397427 y: 8.546694 z: 50.000000
[ INFO] [1583043818.148606818, 15.004000000]: WP: item #2  F:3 C: 16 p: 0.000000 0.000000 0.000000 nan x: 47.397549 y: 8.546050 z: 50.000000
[ INFO] [1583043818.149333410, 15.004000000]: WP: item #3  F:3 C: 16 p: 0.000000 0.000000 0.000000 nan x: 47.398125 y: 8.546157 z: 50.000000
[ INFO] [1583043818.149571936, 15.004000000]: WP: mission received
[ WARN] [1583043818.244105846, 15.100000000]:  Planner abort: missing required data 
 
[ WARN] [1583043818.345181128, 15.200000000]:  Planner abort: missing required data 
 
WARN  [commander] Avoidance system rejected
[ERROR] [1583043818.353060542, 15.208000000]: FCU: Avoidance system rejected
[ WARN] [1583043818.445155253, 15.300000000]:  Planner abort: missing required data 
```

mrivi, 2020-03-02:
It means that you're missing some required input data to run the planner. Check that you there is pointcloud pubished on `/stereo/points2`, a local position on `/mavros/local_position/pose` and check your transformation tree

gitzhangcx, 2020-03-05:
I also have similar problem when i run local planner.Do you solve this problem?

mrivi, 2020-03-16:
have you installed `stereo-image-proc` as written here https://github.com/PX4/avoidance#local-planner-default-heavily-flight-tested ?

reyanshsolis, 2020-05-11:
same issue, and yes I have stereo-image-proc installed. 
ROS-Kinetic, Ubuntu16.04, PX4/avoidance, and Firmware both from the Master branch (on 10th May).