PX4/PX4-Avoidance #565: how do i integrate a lidar sensor to the drone and work on the same stuff? this way, i want to make sure whether the distance being taken into consideration is feasible or not

sanjainn, 2020-03-04:
For general questions please use [PX4 Discuss](http://discuss.px4.io/) or [Slack](http://slack.px4.io/).

**Describe problem solved by the proposed feature**
A clear and concise description of the problem, if any, this feature will solve. E.g. I'm always frustrated when ...

**Describe your preferred solution**
A clear and concise description of what you want to happen.

**Describe possible alternatives**
A clear and concise description of alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots for the feature request here.

sanjainn, 2020-03-04:
this way, i want to make sure whether the distance being taken into consideration is feasible or not. also, how do i add new components in the world scene that is already given.

baumanta, 2020-03-04:
This is not an issue, please follow the guidelines and ask on slack or the forum. If you have a real issue please use the provided template and give a clear description

closed 2020-03-04 by baumanta