PX4/PX4-Avoidance #566: Failed to run local planner ," Planner abort: missing required data "

gitzhangcx, 2020-03-06:
After I run `roslaunch local_planner local_planner_stereo.launch`,I can see the UAV and  world in rviz,
but i can't arm the drone, when i run ｀rosrun mavros mavsys mode -c OFFBOARD｀, it reaponse "Time out". How can i solve it?
`roslaunch local_planner local_planner_stereo.launch
... logging to /home/zhangcx/.ros/log/7aeadb80-5f57-11ea-8442-30e37a24e9dd/roslaunch-zhangcx-HP-Pavilion-Notebook-5367.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://zhangcx-HP-Pavilion-Notebook:44923/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /local_planner_node/goal_x_param: 15
 * /local_planner_node/goal_y_param: 15
 * /local_planner_node/goal_z_param: 4
 * /local_planner_node/pointcloud_topics: ['/stereo/points2']
 * /local_planner_node/world_name: /home/zhangcx/cat...
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://:14540@loca...
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.0003490659 // 0...
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: local_origin
 * /mavros/local_position/tf/child_frame_id: fcu
 * /mavros/local_position/tf/frame_id: local_origin
 * /mavros/local_position/tf/send: True
 * /mavros/local_position/tf/send_fcu: False
 * /mavros/mission/pull_after_gcs: True
 * /mavros/mocap/use_pose: False
[... 285 lines cut ...]
[ WARN] [1583464012.550878583, 15.748000000]:  Planner abort: missing required data 
 
[ WARN] [1583464012.650200162, 15.848000000]:  Planner abort: missing required data 
 
[ WARN] [1583464012.749175830, 15.948000000]:  Planner abort: missing required data 
 
[ WARN] [1583464012.850038388, 16.048000000]:  Planner abort: missing required data 
 
[ WARN] [1583464012.954007388, 16.148000000]:  Planner abort: missing required data 
 
[ WARN] [1583464013.049973058, 16.248000000]:  Planner abort: missing required data 
 
[ WARN] [1583464013.154915106, 16.348000000]:  Planner abort: missing required data 
 
[ INFO] [1583464013.180325072, 16.368000000]: WP: mission received
[ WARN] [1583464013.261705784, 16.448000000]:  Planner abort: missing required data 
 
[ WARN] [1583464013.360228997, 16.548000000]:  Planner abort: missing required data 
 
[ WARN] [1583464013.459154463, 16.648000000]:  Planner abort: missing required data 
 
[ WARN] [1583464013.579661320, 16.748000000]:  Planner abort: missing required data 
 
[ WARN] [1583464013.676663866, 16.848000000]:  Planner abort: missing required data 
`

mrivi, 2020-03-06:
do you see the pointcloud? have you installed the packages in this part of the read me https://github.com/PX4/avoidance#local-planner-default-heavily-flight-tested ?

gitzhangcx, 2020-03-06:
when I run roslaunch `local_planner local_planner_stereo.launch`,i don't see pointcloud.and i have installed the packages. @mrivi

mrivi, 2020-03-16:
can you please double check that you have the stereo-image-proc package? That's the only difference with respect to the other launch file that it's working for you

gitzhangcx, 2020-03-17:
I delete all model files in the folder `~/.gazebo/model`, and reinstall ROS and Gazebo. I can run both stereo and depth launch files successfully. May be the difference between local camera models and the database models results in this error?

mrivi, 2020-03-17:
They are in `Firmware/Tools/sitl_gazebo`

closed 2020-03-17 by mrivi

rk800506, 2021-07-23:
@mrivi Is there any way to solve the issue without reinstalling ros and gazebo

Long-Wu-code, 2021-12-06:
> @mrivi Is there any way to solve the issue without reinstalling ros and gazebo

You can try copy all models in Firmware/Tools/sitl_gazebo/models,  and then paste all of them in .gazebo/models. Let new models replace original one!

Aaron1356, 2023-11-06:
So I tried this approach to solving the issue but it did not work for me. I tried Replacing the Models in Gazebo, and I tried uninstalling ROS and Gazebo from the system. Still getting the same warning that Data is missing and there is no image appearing in the image window of Local Planner