PX4/PX4-Avoidance #567: rosrun mavros mavsys mode -c OFFBOARD

150202029, 2020-03-08:
Hello
The error I get when I run the "rosrun mavros mavsys mode -c OFFBOARD" and "rosrun mavros mavsafety arm" commands below. "roslaunch global_planner global_planner_stereo.launch"

[image]
 command is running before running this command

mrivi, 2020-03-09:
please share the console log from `roslaunch global_planner global_planner_stereo.launch`

150202029, 2020-03-14:
nevrize@nevrizenergis:~/catkin_ws$ roslaunch global_planner global_planner_stereo.launch
... logging to /home/nevrize/.ros/log/73599ed0-65c7-11ea-b732-50eb71ccc82e/roslaunch-nevrizenergis-6000.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:37503/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /global_planner_node/frame_id: local_origin
 * /global_planner_node/pointcloud_topics: /camera/depth/points
 * /global_planner_node/robot_radius: 0.5
 * /global_planner_node/start_pos_x: 0.5
 * /global_planner_node/start_pos_y: 0.5
 * /global_planner_node/start_pos_z: 3.5
 * /global_planner_node/world_name: /home/nevrize/cat...
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://:14540@loca...
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.0003490659 // 0...
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: local_origin
 * /mavros/local_position/tf/child_frame_id: fcu
 * /mavros/local_position/tf/frame_id: local_origin
 * /mavros/local_position/tf/send: True
 * /mavros/local_position/tf/send_fcu: False
 * /mavros/mission/pull_after_gcs: True
 * /mavros/mocap/use_pose: False
[... 274 lines cut ...]
[ERROR] [1584171796.430800666]: PR: Unknown parameter to get: MPC_XY_CRUISE
[ERROR] [1584171796.432529617]: PR: Unknown parameter to get: CP_DIST
[ERROR] [1584171796.434271294]: PR: Unknown parameter to get: MPC_LAND_SPEED
[ERROR] [1584171796.436117515]: PR: Unknown parameter to get: MPC_JERK_MAX
[ERROR] [1584171796.437625507]: PR: Unknown parameter to get: NAV_ACC_RAD
[ERROR] [1584171801.439703743]: PR: Unknown parameter to get: MPC_ACC_HOR
[ERROR] [1584171801.442045209]: PR: Unknown parameter to get: MPC_XY_CRUISE
[ERROR] [1584171801.444804361]: PR: Unknown parameter to get: CP_DIST
[ERROR] [1584171801.448264615]: PR: Unknown parameter to get: MPC_LAND_SPEED
[ERROR] [1584171801.451858499]: PR: Unknown parameter to get: MPC_JERK_MAX
[ERROR] [1584171801.453257037]: PR: Unknown parameter to get: NAV_ACC_RAD
[ERROR] [1584171806.456120968]: PR: Unknown parameter to get: MPC_ACC_HOR
[ERROR] [1584171806.458030926]: PR: Unknown parameter to get: MPC_XY_CRUISE
[ERROR] [1584171806.459312111]: PR: Unknown parameter to get: CP_DIST
[ERROR] [1584171806.460752362]: PR: Unknown parameter to get: MPC_LAND_SPEED
[ERROR] [1584171806.462135789]: PR: Unknown parameter to get: MPC_JERK_MAX
[ERROR] [1584171806.463915539]: PR: Unknown parameter to get: NAV_ACC_RAD
[ERROR] [1584171811.466738051]: PR: Unknown parameter to get: MPC_ACC_HOR
[ERROR] [1584171811.468383032]: PR: Unknown parameter to get: MPC_XY_CRUISE
[ERROR] [1584171811.470100800]: PR: Unknown parameter to get: CP_DIST
[ERROR] [1584171811.472044155]: PR: Unknown parameter to get: MPC_LAND_SPEED
[ERROR] [1584171811.473906106]: PR: Unknown parameter to get: MPC_JERK_MAX
[ERROR] [1584171811.475609253]: PR: Unknown parameter to get: NAV_ACC_RAD
[ERROR] [1584171816.477680012]: PR: Unknown parameter to get: MPC_ACC_HOR
[ERROR] [1584171816.479697403]: PR: Unknown parameter to get: MPC_XY_CRUISE

mrivi, 2020-03-16:
I need more information. Are you receiving `/mavros/local_position/pose` and `/stereo/points2`?

150202029, 2020-03-17:
nevrize@nevrizenergis:~$ rostopic list
/actual_path
/blocked_marker
/camera/rgb/image_raw
/clicked_point
/clock
/closest_point
/closest_point_array
/cloud_in
/current_setpoint
/current_setpoint_array
/deg60_point
/deg60_point_array
/diagnostics
/drone
/drone_array
/explored_cells
/extended_marker
/free_cells_vis_array
/global_goal
/global_planner_node/parameter_descriptions
/global_planner_node/parameter_updates
/global_temp_goal
/global_temp_path
/goal_position
/local_pointcloud
/mavlink/from
/mavlink/to
/mavros/actuator_control
/mavros/adsb/send
/mavros/adsb/vehicle
/mavros/altitude
/mavros/battery
/mavros/cam_imu_sync/cam_imu_stamp
/mavros/companion_process/status
/mavros/debug_value/debug
/mavros/debug_value/debug_vector
/mavros/debug_value/named_value_float
/mavros/debug_value/named_value_int
/mavros/debug_value/send
/mavros/estimator_status
/mavros/extended_state
/mavros/fake_gps/mocap/tf
/mavros/global_position/compass_hdg
/mavros/global_position/global
/mavros/global_position/gp_lp_offset
/mavros/global_position/gp_origin
/mavros/global_position/home
/mavros/global_position/local
/mavros/global_position/raw/fix
/mavros/global_position/raw/gps_vel
/mavros/global_position/raw/satellites
/mavros/global_position/rel_alt
/mavros/global_position/set_gp_origin
/mavros/gps_rtk/send_rtcm
/mavros/hil/actuator_controls
/mavros/hil/controls
/mavros/hil/gps
/mavros/hil/imu_ned
/mavros/hil/optical_flow
/mavros/hil/rc_inputs
/mavros/hil/state
/mavros/home_position/home
/mavros/home_position/set
/mavros/imu/data
/mavros/imu/data_raw
/mavros/imu/diff_pressure
/mavros/imu/mag
/mavros/imu/static_pressure
/mavros/imu/temperature_baro
[... 157 lines cut ...]
/stereo/right/image_rect_color/compressed/parameter_descriptions
/stereo/right/image_rect_color/compressed/parameter_updates
/stereo/right/image_rect_color/compressedDepth
/stereo/right/image_rect_color/compressedDepth/parameter_descriptions
/stereo/right/image_rect_color/compressedDepth/parameter_updates
/stereo/right/image_rect_color/theora
/stereo/right/image_rect_color/theora/parameter_descriptions
/stereo/right/image_rect_color/theora/parameter_updates
/stereo/stereo_image_proc/parameter_descriptions
/stereo/stereo_image_proc/parameter_updates
/stereo/stereo_image_proc_debayer_left/parameter_descriptions
/stereo/stereo_image_proc_debayer_left/parameter_updates
/stereo/stereo_image_proc_debayer_right/parameter_descriptions
/stereo/stereo_image_proc_debayer_right/parameter_updates
/stereo/stereo_image_proc_rectify_color_left/parameter_descriptions
/stereo/stereo_image_proc_rectify_color_left/parameter_updates
/stereo/stereo_image_proc_rectify_color_right/parameter_descriptions
/stereo/stereo_image_proc_rectify_color_right/parameter_updates
/stereo/stereo_image_proc_rectify_mono_left/parameter_descriptions
/stereo/stereo_image_proc_rectify_mono_left/parameter_updates
/stereo/stereo_image_proc_rectify_mono_right/parameter_descriptions
/stereo/stereo_image_proc_rectify_mono_right/parameter_updates
/tf
/tf_static
/world

150202029, 2020-03-17:
nevrize@nevrizenergis:~$ rosservice list
/gazebo_gui/get_loggers
/gazebo_gui/set_logger_level
/global_planner_node/get_loggers
/global_planner_node/set_logger_level
/global_planner_node/set_parameters
/mavros/cmd/arming
/mavros/cmd/command
/mavros/cmd/command_int
/mavros/cmd/land
/mavros/cmd/set_home
/mavros/cmd/takeoff
/mavros/cmd/trigger_control
/mavros/cmd/trigger_interval
/mavros/cmd/vtol_transition
/mavros/ftp/checksum
/mavros/ftp/close
/mavros/ftp/list
/mavros/ftp/mkdir
/mavros/ftp/open
/mavros/ftp/read
/mavros/ftp/remove
/mavros/ftp/rename
/mavros/ftp/reset
/mavros/ftp/rmdir
/mavros/ftp/truncate
/mavros/ftp/write
/mavros/get_loggers
/mavros/home_position/req_update
/mavros/log_transfer/raw/log_request_data
/mavros/log_transfer/raw/log_request_end
/mavros/log_transfer/raw/log_request_list
/mavros/mission/clear
/mavros/mission/pull
/mavros/mission/push
/mavros/mission/set_current
/mavros/mount_control/configure
/mavros/param/get
/mavros/param/pull
/mavros/param/push
/mavros/param/set
/mavros/set_logger_level
/mavros/set_message_interval
/mavros/set_mode
/mavros/set_stream_rate
/mavros/setpoint_position/mav_frame
/mavros/setpoint_trajectory/mav_frame
/mavros/setpoint_trajectory/reset
/mavros/setpoint_velocity/mav_frame
/mavros/vehicle_info_get
/octomap_binary
/octomap_full
/octomap_server/clear_bbx
/octomap_server/get_loggers
/octomap_server/reset
/octomap_server/set_logger_level
/octomap_server/set_parameters
/rosout/get_loggers
/rosout/set_logger_level
/rqt_gui_py_node_20758/get_loggers
/rqt_gui_py_node_20758/set_logger_level
/rqt_reconfigure/get_loggers
/rqt_reconfigure/set_logger_level
/rviz/get_loggers
/rviz/reload_shaders
/rviz/set_logger_level
/spawn_model/get_loggers
/spawn_model/set_logger_level
/stereo/left/image_color/compressed/set_parameters
/stereo/left/image_color/compressedDepth/set_parameters
/stereo/left/image_color/theora/set_parameters
/stereo/left/image_mono/compressed/set_parameters
/stereo/left/image_mono/compressedDepth/set_parameters
/stereo/left/image_mono/theora/set_parameters
/stereo/left/image_rect/compressed/set_parameters
/stereo/left/image_rect/compressedDepth/set_parameters
/stereo/left/image_rect/theora/set_parameters
/stereo/left/image_rect_color/compressed/set_parameters
/stereo/left/image_rect_color/compressedDepth/set_parameters
/stereo/left/image_rect_color/theora/set_parameters
/stereo/right/image_color/compressed/set_parameters
/stereo/right/image_color/compressedDepth/set_parameters
/stereo/right/image_color/theora/set_parameters
/stereo/right/image_mono/compressed/set_parameters
/stereo/right/image_mono/compressedDepth/set_parameters
/stereo/right/image_mono/theora/set_parameters
/stereo/right/image_rect/compressed/set_parameters
/stereo/right/image_rect/compressedDepth/set_parameters
/stereo/right/image_rect/theora/set_parameters
/stereo/right/image_rect_color/compressed/set_parameters
/stereo/right/image_rect_color/compressedDepth/set_parameters
/stereo/right/image_rect_color/theora/set_parameters
/stereo/stereo_image_proc/get_loggers
/stereo/stereo_image_proc/set_logger_level
/stereo/stereo_image_proc/set_parameters
/stereo/stereo_image_proc_debayer_left/set_parameters
/stereo/stereo_image_proc_debayer_right/set_parameters
/stereo/stereo_image_proc_rectify_color_left/set_parameters
/stereo/stereo_image_proc_rectify_color_right/set_parameters
/stereo/stereo_image_proc_rectify_mono_left/set_parameters
/stereo/stereo_image_proc_rectify_mono_right/set_parameters
/tf_camera/get_loggers
/tf_camera/set_logger_level
/tf_depth_camera/get_loggers
/tf_depth_camera/set_logger_level
/tf_local_origin/get_loggers
/tf_local_origin/set_logger_level

mrivi, 2020-03-17:
You need to do `rosotpic hz /topic_name` to actually see if any message is being received

150202029, 2020-03-17:
hi, its new log

[ WARN] [1584482485.937315769, 173.400000000]:  Planner abort: missing required data 
 
[ WARN] [1584482486.027750561, 173.500000000]:  Planner abort: missing required data 
 
[ WARN] [1584482486.132736951, 173.600000000]:  Planner abort: missing required data 
 
[ WARN] [1584482486.236411396, 173.700000000]:  Planner abort: missing required data 
 
[ERROR] [1584482486.307740472, 173.772000000]: PR: Unknown parameter to get: MPC_ACC_HOR
[ERROR] [1584482486.321268743, 173.784000000]: PR: Unknown parameter to get: MPC_XY_CRUISE
[ERROR] [1584482486.331530481, 173.796000000]: PR: Unknown parameter to get: CP_DIST
[ WARN] [1584482486.335417301, 173.800000000]:  Planner abort: missing required data 
 
[ERROR] [1584482486.340402484, 173.804000000]: PR: Unknown parameter to get: MPC_LAND_SPEED
[ERROR] [1584482486.349775112, 173.816000000]: PR: Unknown parameter to get: MPC_JERK_MAX
[ERROR] [1584482486.357386643, 173.820000000]: PR: Unknown parameter to get: NAV_ACC_RAD
[ WARN] [1584482486.443369929, 173.904000000]:  Planner abort: missing required data 
 
[ INFO] [1584482486.539288257, 174.000000000]:   STOP  
[ INFO] [1584482486.539590912, 174.000000000]: Goal position is occupied
[ INFO] [1584482486.539701305, 174.000000000]: Failed to find a path
Total time: 0.46 ms 
[ WARN] [1584482486.539835017, 174.000000000]:  Planner abort: missing required data 
 
[ INFO] [1584482486.540262318, 174.000000000]:   Received empty path

[ INFO] [1584482486.540542943, 174.000000000]:   Received empty path

[ WARN] [1584482486.641935776, 174.100000000]:  Planner abort: missing required data 
 
[ WARN] [1584482486.748122123, 174.200000000]:  Planner abort: missing required data 
 
[ WARN] [1584482486.849393582, 174.300000000]:  Planner abort: missing required data 
 
[ WARN] [1584482486.949315147, 174.400000000]:  Planner abort: missing required data 
 
[ WARN] [1584482487.044687016, 174.500000000]:  Planner abort: missing required data 
 
[ WARN] [1584482487.145415691, 174.600000000]:  Planner abort: missing required data 
 
[ WARN] [1584482487.240512202, 174.700000000]:  Planner abort: missing required data 
 
[ WARN] [1584482487.337148521, 174.800000000]:  Planner abort: missing required data 
 
[ WARN] [1584482487.438689680, 174.900000000]:  Planner abort: missing required data 
 
[ INFO] [1584482487.543366385, 175.000000000]:   STOP  
[ INFO] [1584482487.543715889, 175.000000000]: Goal position is occupied
[ INFO] [1584482487.543809052, 175.000000000]: Failed to find a path
Total time: 0.40 ms 
[ INFO] [1584482487.543996667, 175.000000000]:   Received empty path

[ INFO] [1584482487.544060486, 175.000000000]:   Received empty path

[ WARN] [1584482487.544225995, 175.000000000]:  Planner abort: missing required data 
 
[ WARN] [1584482487.676255944, 175.108000000]:  Planner abort: missing required data 
 
[ WARN] [1584482487.775462959, 175.200000000]:  Planner abort: missing required data 
 
[ WARN] [1584482487.876434033, 175.300000000]:  Planner abort: missing required data 
 
[ WARN] [1584482487.982145789, 175.400000000]:  Planner abort: missing required data 
 
[ WARN] [1584482488.087373918, 175.504000000]:  Planner abort: missing required data 
 
[ WARN] [1584482488.187332356, 175.600000000]:  Planner abort: missing required data 
 
[... 82 lines cut ...]
[ WARN] [1584482491.205957601, 178.600000000]:  Planner abort: missing required data 
 
[ WARN] [1584482491.308217296, 178.700000000]:  Planner abort: missing required data 
 
[ERROR] [1584482491.365138514, 178.760000000]: PR: Unknown parameter to get: MPC_ACC_HOR
[ERROR] [1584482491.373196230, 178.768000000]: PR: Unknown parameter to get: MPC_XY_CRUISE
[ERROR] [1584482491.386278141, 178.772000000]: PR: Unknown parameter to get: CP_DIST
[ERROR] [1584482491.394706745, 178.792000000]: PR: Unknown parameter to get: MPC_LAND_SPEED
[ERROR] [1584482491.400220981, 178.796000000]: PR: Unknown parameter to get: MPC_JERK_MAX
[ WARN] [1584482491.404024316, 178.800000000]:  Planner abort: missing required data 
 
[ERROR] [1584482491.405347964, 178.800000000]: PR: Unknown parameter to get: NAV_ACC_RAD
[ WARN] [1584482491.502153681, 178.900000000]:  Planner abort: missing required data 
 
[ INFO] [1584482491.606801299, 179.000000000]:   STOP  
[ WARN] [1584482491.606975436, 179.000000000]:  Planner abort: missing required data 
 
[ INFO] [1584482491.607391425, 179.000000000]: Goal position is occupied
[ INFO] [1584482491.607453942, 179.000000000]: Failed to find a path
Total time: 1.39 ms 
[ INFO] [1584482491.607592682, 179.000000000]:   Received empty path

[ INFO] [1584482491.607677317, 179.000000000]:   Received empty path

[ WARN] [1584482491.707574349, 179.100000000]:  Planner abort: missing required data

150202029, 2020-03-17:
rostopic hz /mavros/local_position/pose
subscribed to [/mavros/local_position/pose]
WARNING: may be using simulated time
no new messages
no new messages
no new messages
no new messages
no new messages

rostopic hz /stereo/points2
subscribed to [/stereo/points2]
WARNING: may be using simulated time
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
^Cno new messages

mrivi, 2020-03-23:
so you aren't receiving the vehicle position from the autopilot not the simulated pointcloud from gazebo. There is something wrong with how you have set up your environment. Please, go back to the readme and follow the instructions again.

150202029, 2020-03-24:
I followed the installation steps completely but the result was the same. which step do you think the error might have originated from?

LLeeSG, 2021-07-23:
mind ur PX4_SIM_HOST set