PX4/PX4-Avoidance #568: [Local Planner] Avoidance system not ready

SwiftGust, 2020-03-12:
**Describe the bug**
A clear and concise description of what the bug is.
With current master of avoidance, Firmware cannot able to arm
even mavros/companion_status/status message is updated well as
state = 4 (MAV_STATE_ACTIVE)
component = 196 (MAV_COMP_ID_OBSTACLE_AVOIDANCE)

Firmware - built 02.18
mavlink-gbp-release(https://github.com/mavlink/mavlink-gbp-release/tree/upstream/2020.2.2

**To Reproduce**
Steps to reproduce the behavior:
1. Drone switched on 
2. Set parameter COM_OBS_AVOID = 1 & reboot
3. Try to arm
4. Prearm check fails with Avoidance system not ready

**Expected behavior**
A clear and concise description of what you expected to happen.
Vehicle should be able to arm without error
**Log Files and Screenshots**
*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/en/getting_started/flight_reporting.html)) or from the Gazebo simulation (logs are located in `~/.ros/logs/`)
- Share the link to a log showing the problem on [PX4 Flight Review](http://logs.px4.io/).
- Share a link to a rosbag showing the problem

Jaeyoung-Lim, 2020-03-12:
@SwiftGust Which firmware version are you running?

SwiftGust, 2020-03-13:
@Jaeyoung-Lim No problem found in Firmware, issue found in mavros.
The problem is found from the Firmware is the mavros was not setting system id set in CompanionProcessStatus Plugin.

The vehicle i was using was set to sys_id =11 (PX4) but heartbeat generated form CompanionProcessStatus plugins found to be set to 1 and the Firmware checks system id coming in except it is GCS heartbeat.

Jaeyoung-Lim, 2020-03-13:
@SwiftGust How did you solve the problem?

SwiftGust, 2020-03-13:
Since heartbeat mavlink message definition does not have system_id field,
just set to vehicle sys_id to 1 for the quick fix.
https://mavlink.io/en/messages/common.html#HEARTBEAT

Did not test using mavros hidden system_id parameter yet
https://github.com/mavlink/mavros/blob/dcef73674718a80202d68336cb0aa6e88142a71d/mavros/src/lib/mavros.cpp#L56 &
https://github.com/mavlink/c_library_v2/blob/e8bdbcfbacccd79fc678f5c6755c8fb680021199/common/mavlink_msg_heartbeat.h#L92