PX4/PX4-Avoidance #569: 3DVFH algorithm for local planner?

gitzhangcx, 2020-03-25:
which algorithm does the local_planner  use, 3DVFH or 3DVFH*?
how can i evaluate the advantages of the 3DVFH* algorithm over the 3DVFH algorithm? How to run the two algorithms separately? @mrivi

baumanta, 2020-03-25:
the local_planner runs 3DVFH*. You can see in SITL that it constructs a tree of possible paths and then determines the best one to follow. If you would use 3DVFH you would only get the first edge (no tree). It is not possible to run 3DVFH with the current code. But if you are looking for a comparison of the algorithms, this might help you: https://drive.google.com/file/d/1yjDtxRrIntr5Mdaj9CCB4IFJn0Iy2-bR/view

Norton-z, 2020-03-25:
Thanks a lot! 
By the way, how can i set a fixed goal in Rviz when i run local planner? because i want to do several simulation experiments to analyze the trajectory of the uav.

baumanta, 2020-03-25:
The local_planner is mean to be used integrated with the PX4 firmware. You can use QGC to plan missions and the waypoints will be automatically transferred to the planner as goals.

For the future, please use the forum discuss.px4.io/ or slack for questions.This page is only if you would like to report issues with the code

closed 2020-03-25 by baumanta