PX4/PX4-Avoidance #570: where is safety margin variable in local_planner?

Norton-z, 2020-03-31:
i want to know the value of safety margin in local_planner?
where is the safety margin variable defined?
@mrivi @baumanta

jkflying, 2020-03-31:
It gets the value from the firmware for CP_DIST

closed 2020-03-31 by jkflying

Norton-z, 2020-03-31:
what is CP_DIST? can i find the parameters in the node?

baumanta, 2020-03-31:
CP_DIST is for collision prevention only (position flight mode) as far as I can remember. If you want to keep a certain distance during a mission, you would need to tweek this one here: https://github.com/PX4/avoidance/blob/c0c159bb6d260e323a9140c8f277daf6db2bdecb/local_planner/cfg/LocalPlannerNode.cfg#L15
But be careful it is one on many cost terms, so some value sets might not produce good results.

For the future, this is not a support forum. Please ask questions on the forum of slack