PX4/PX4-Avoidance #571: issue in using px4/avoidance/local_planner local_planner_depth-camera.launch

rrajnidhi, 2020-03-31:
Hi,
I am trying to do simulation, i faced with following issues:
1. Even on mission mode after connecting with qgcs, i couldnot see any uav mode in local_planner.rviz.(attaching screenshot)
2. Also during mission obstacles are not avoiding, uav is getting crashed every single time to an obstacle.! seems like local_planner is not working .!!
3. In doc i have read COM_OBS_AVOID should be set but seting to 1 gives me an error in qgcs saying aircraft is not ready for flight. so i am flying it with COM_OBS_AVOID set to 0. ! is it a proper way.?

[image]

mrivi, 2020-04-01:
You need to have COM_OBS_AVOID set to 1 otherwise Obstacle Avoidance is disabled and you will fly into obstacles. In Rviz you should see the vehicle, the world and the pointcloud. Most liklely your environment is not set correctly. Please run the following commands and share the output here
```bash
echo $ROS_PACKAGE_PATH
echo $GAZEBO_PLUGIN_PATH
echo $LD_LIBRARY_PATH
```

rrajnidhi, 2020-04-01:
Hi, attaching the outputs for the commands :
nidhi@nidhi-HP:~$ echo $ROS_PACKAGE_PATH
/home/nidhi/catkin_ws/src/beg_tut:/home/nidhi/catkin_ws/src/mavlink:/home/nidhi/catkin_ws/src/mavros/libmavconn:/home/nidhi/catkin_ws/src/mavros/mavros_msgs:/home/nidhi/catkin_ws/src/mavros/mavros:/home/nidhi/catkin_ws/src/mavros/mavros_extras:/home/nidhi/catkin_ws/src/avoidance/avoidance:/home/nidhi/catkin_ws/src/avoidance/global_planner:/home/nidhi/catkin_ws/src/avoidance/local_planner:/home/nidhi/catkin_ws/src/avoidance/safe_landing_planner:/home/nidhi/catkin_ws/src/mavros/test_mavros:/opt/ros/kinetic/share
nidhi@nidhi-HP:~$ echo $GAZEBO_PLUGIN_PATH

nidhi@nidhi-HP:~$ echo $GAZEBO_MODEL_PATH
:/home/nidhi/Firmware/Tools/sitl_gazebo/models:/home/nidhi/catkin_ws/src/avoidance/avoidance/sim/models:/home/nidhi/catkin_ws/src/avoidance/avoidance/sim/worlds
nidhi@nidhi-HP:~$ echo $LD_LIBRARY_PATH
/home/nidhi/catkin_ws/devel/lib:/opt/ros/kinetic/lib:/opt/ros/kinetic/lib/x86_64-linux-gnu

mrivi, 2020-04-02:
most likely you haven;t run the instructions at point 3 here https://github.com/PX4/avoidance#build-and-run-the-simulator

Make sure to change the last line based on the path to the Firmware directory on your machine