PX4/PX4-Avoidance #575: rosrun mavros mavsys mode -c OFFBOARD

pkyadav73199, 2020-05-03:
**Describe the bug**
when i am trying to start offboard mission by runing rosrun mavros mavsys mode -c OFFBOARD , i am getting error [ERROR] [1588485064.541397576]: MODE: Unsupported FCU
Error [parser.cc:689] Unable to find uri[model://iris_depth_camera]

pkyadav73199, 2020-05-03:
i am also getting this 

edith@edith:~/catkin_ws$ rosrun mavros mavsys mode -c OFFBOARD
Request failed. Check mavros logs

edith@edith:~/catkin_ws$ rosrun mavros mavsafety arm
Request failed. Check mavros logs

Jaeyoung-Lim, 2020-05-03:
```
 Error [parser.cc:689] Unable to find uri[model://iris_depth_camera]
```
This output is not related to your command. It is coming because you didn't set the `GAZEBO_MODEL_PATH` properly. 
Run
```
cd <Firmware_clone>
source ~/catkin_ws/devel/setup.bash
source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)/Tools/sitl_gazebo
```

pkyadav73199, 2020-05-04:
Thanks a lot.
It  works for me.

closed 2020-05-04 by pkyadav73199