PX4/PX4-Avoidance #576: Global planner doesn't allow to fly below 2 meters

Sebboo97, 2020-05-09:
**Describe the bug**
Whenever I set a clicked goal altitude below 2 meters, the global planner won't fly there and says "goal position is occupied"

**To Reproduce**
Steps to reproduce the behavior:
1. Drone is in Offboard mode
2. Setting goal altitude to 1.8m and min_altitude to 0m in Dynamic Reconfigure, see screenshot
[image]

3. Click goal in Rviz
4. See error and log file below
[ INFO] [1589013752.459053907, 34.960000000]: OctoMap memory usage: 0.046 MB
[ INFO] [1589013752.459185862, 34.960000000]: Planning a path from (5,-11,3) to (5,-11,3)
[ INFO] [1589013752.459213565, 34.960000000]: curr_pos_: 5.51,-10.69,3.57	 s: 5.50,-10.50,3.50
Search              iter_time overest   num_iter  path_cost 
NodeWithoutSmooth   inf       2         0         0         (cost: 0.00, dist: 0.00, risk: 0.00, smooth: 0.00) 
SpeedNode           inf       1.25      0         0         (cost: 0.00, dist: 0.00, risk: 0.00, smooth: 0.00) 
SpeedNode           -nan      1.06      0         0         (cost: 0.00, dist: 0.00, risk: 0.00, smooth: 0.00) 
Total time: 0.77 ms 
[ INFO] [1589013753.137902542, 35.580000000]: Setting goal: Frame:local_origin, Position(0.530, -18.315, 0.000), Orientation(0.000, 0.000, -0.447, 0.895) = Angle: -0.927
[ INFO] [1589013753.217684820, 35.656000000]: ========== Set goal : (0,-19,1) ==========
[ INFO] [1589013753.543816882, 35.964000000]: OctoMap memory usage: 0.046 MB
[ INFO] [1589013753.543924621, 35.964000000]: Goal position is occupied
[ INFO] [1589013753.543989295, 35.964000000]: Failed to find a path
Total time: 0.43 ms 
[ INFO] [1589013754.571964586, 36.960000000]:   STOP  
[ INFO] [1589013754.572157098, 36.960000000]: OctoMap memory usage: 0.046 MB
[ INFO] [1589013754.572251670, 36.960000000]: Planning a path from (5,-11,3) to (5,-11,3)
[ INFO] [1589013754.572324511, 36.960000000]: curr_pos_: 5.53,-10.64,3.57	 s: 5.50,-10.50,3.50
Search              iter_time overest   num_iter  path_cost 
NodeWithoutSmooth   inf       2         0         0         (cost: 0.00, dist: 0.00, risk: 0.00, smooth: 0.00) 
SpeedNode           inf       1.25      0         0         (cost: 0.00, dist: 0.00, risk: 0.00, smooth: 0.00) 
SpeedNode           inf       1.06      0         0         (cost: 0.00, dist: 0.00, risk: 0.00, smooth: 0.00) 
Total time: 1.02 ms

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
[https://logs.px4.io/plot_app?log=86bfb4f7-a721-456d-87d9-12341c47af7a](url)

**Additional context**
Is this the expected behaviour?
If yes, why is the local planner able to fly below 2m but the global is not?
If no, any ideas what could be the problem?
I also wonder, why the "min_altitude" setting does not have any effect on the planner.
If I set it to 5m, I still can fly below, e.g. to 3m. And setting it to 0m won't make it possible to fly below 2 m.

Jaeyoung-Lim, 2020-05-22:
@Sebboo97 Yes, this is the expected behavior, the map is trying to avoid taking the ground as part of the obstacle map.

Sebboo97, 2020-05-22:
@Jaeyoung-Lim Thank you for your answer!
I have now changed _max_cell_risk_ to >1,1 in the code and with that I'm able to fly below 2m without any problems.
Is there a reason why _max_cell_risk_ is limited to 1 by default in the _rqt_reconfigure_ tool?
In other words, may it cause a problem to leave it at 1,1?