PX4/PX4-Avoidance #577: Local planner stereo launch file: system power unavailable and time out

MagnusCardell, 2020-05-11:
**Describe the bug**
A clear and concise description of what the bug is.
`roslaunch local_planner local_planner_stereo.launch` starts px4 local planner but subsequent `rosrun mavros mavsys mode -c OFFBOARD ` results in `Timed out!`. 

I have tried to look through the output but did not find anything apparent wrong. Reading through the issues here also lead me to trying to start by taking of from the QGC. However when I do, I get a `system power unavailable` response.

I tried pulling the latest from Firmware and rebuild but no difference. Could this issue: https://github.com/PX4/avoidance/issues/536 be related? 

I am attaching the output in a file. 

**To Reproduce**
Steps to reproduce the behavior:
1. roslaunch local_planner local_planner_stereo.launch
2. rosrun mavros mavsys mode -c OFFBOARD (in new terminal) - ERROR
3. Or takeoff/arm from QGC - ERROR

**Expected behavior**
Offboard mode activated, able to arm, and run the local_planner node. 

**Log Files and Screenshots**
Log file attached

[tmux2.log](https://github.com/PX4/avoidance/files/4608695/tmux2.log)

MagnusCardell, 2020-05-11:
Update: Cloned a fresh copy of the Avoidance repo to build and run the same way. Same issue.

mrivi, 2020-05-15:
From the log it seems that you are running safe landing planner. 
Furthermore `WARN  [PreFlightCheck] system power unavailable` isn't related to an avodance issue, it's flight control. 
Can you run `make px4_sitl gazebo` and fly?

atharva417, 2020-07-29:
yes

atharva417, 2020-07-29:
I can run make px4_sitl gazebo and can fly iris using commander takeoff. 
But when I launch using mavros_posix_sitl.launch and type commander takeoff it gives error system power unavailable.