PX4/PX4-Avoidance #578: TF Buffer: could not retrieve requested transform from buffer, unregistered. Drone not showing

AldoCF96, 2020-05-15:
**Describe the bug**
TF Buffer: could not retrieve requested transform from buffer, unregistered. Drone not showing.

**To Reproduce**
Steps to reproduce the behavior:
run simulation, close the simulation.

 run  roslaunch local_planner local_planner_stereo.launch but the drone never shows and it display message error as the tittle.

**Expected behavior**
exptect to show the  drone, and be able to connect to the drone

[image]

mrivi, 2020-05-15:
You probably haven't set up correctly the environment. Could you please share the outputs of 
```bash
echo $ROS_PACKAGE_PATH
echo $GAZEBO_MODEL_PATH
echo $GAZEBO_PLUGIN_PATH
echo $LD_LIBRARY_PATH

AldoCF96, 2020-05-15:
[image]

Actually I never set up the plugin path it wasnt on the instructions

mrivi, 2020-05-15:
You haven't run point 3 from https://github.com/PX4/avoidance#build-and-run-the-simulator correctly. Please run it again. The script `setup_gazebo.bash` sets those paths. Make sure to modify the paths to the Firmware to match your system

AldoCF96, 2020-05-15:
I run it and I got this: 
aldo@aldo-TP501UB:~/Firmware$ echo $GAZEBO_PLUGIN_PATH
:/home/aldo/Firmware/build/px4_sitl_default/build_gazebo

It still shows the same error.  can you expand the part about modifying the paths ?

AldoCF96, 2020-05-15:
I pulled the latest version of the code. is there something that I am missing ?

AldoCF96, 2020-05-16:
FIxed: Apparently in Ubuntu 16.04 even when  it shows the PATH has been set up , you need to set up again, And run everything on the same terminal if you change terminal it wont work. 
 Need to test if it is better to sets all does paths with a screen and  run it from terminal seems too much work to  repeat these process all over again.

closed 2020-05-16 by AldoCF96