PX4/PX4-Avoidance #582: [Local Planner] Can't see buildings in rviz when launched with test_city_2.world

ytw1996, 2020-05-22:
**Describe the bug**
After I modified the local_planner_stereo.launch file with the replacement of 

> simple_obstacle

 in line 3 with 

> test_city_2

, I run `roslaunch local_planner local_planner_stereo.launch` in a terminal and can't see buildings in the rviz. I can see the drone model with some red feature points it detects as shown in the figure
[image]
However, I can launch the default local_planner_stereo.launch file without trouble and the drone can be armed and navigate to points I set.
I am attaching the output in a file.
**Expected behavior**
In the rviz, the buildings are displayed.
Log Files and Screenshots
[terminal.log](https://github.com/PX4/avoidance/files/4665803/terminal.log)

nicovanduijn, 2020-05-22:
the visualization in RViz is not automatically tied to Gazebo. It's a pretty hacky implementation that loads a (manually created) yaml file that places the objects accordingly...

nicovanduijn, 2020-05-22:
Basically, you need to create `test_city_2.yaml` by hand if you want this to work

closed 2020-05-22 by nicovanduijn

ytw1996, 2020-05-23:
@nicovanduijn Thanks for your replying

pkyadav73199, 2020-06-05:
@nicovanduijn  is there any other way to launch test_city_2.world file in rviz without the use of ymal file ??

baumanta, 2020-06-08:
@pkyadav73199 No. Not if you want the world to show in rviz.