PX4/PX4-Avoidance #583: Can not start simulation with  "roslaunch local_planner local_planner_depth-camera.launch"

Cristian-wp, 2020-05-25:
Hello,
I am trying to run the simulation following your tutorial, but i can not make it run.
When I run the command "roslaunch local_planner local_planner_depth-camera.launch" I get the following outputs reported in Node logs.
As you can see RViz show this error "Fixed Frame [local_origin] does not exist" and the terminal show errors like it can not read the parameter of the simulated PX4.
I have put the Firmware folder inside the src folder and I have modify the instruction as reported in the guide but it does not work.
I have install mavros as you say, but I had to put the folder even inside my src, otherwise i get compilation error about the missing package.
Can you please help me to run the simulation? I am new to PX4 and ROS, so sorry if is a dummy question :)

### MAVROS version and platform

Mavros: 1.2.0
ROS: Melodic
Ubuntu: 18.04

### Autopilot type and version

[  ] ArduPilot
[ x ] PX4

Version: v1.10.1

### Node logs

[image]

```
SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /local_planner_nodelet/goal_x_param: 17
 * /local_planner_nodelet/goal_y_param: 15
 * /local_planner_nodelet/goal_z_param: 3
 * /local_planner_nodelet/pointcloud_topics: ['/camera/depth/p...
 * /local_planner_nodelet/world_name: /home/sei/catkin_...
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://:14540@loca...
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
[... 322 lines cut ...]
[ERROR] [1590423450.485496642]: PR: Unknown parameter to get: MPC_YAWRAUTO_MAX
[ERROR] [1590423455.487711458]: PR: Unknown parameter to get: MPC_ACC_HOR
[ERROR] [1590423455.489656304]: PR: Unknown parameter to get: MPC_ACC_DOWN_MAX
[ERROR] [1590423455.491516339]: PR: Unknown parameter to get: MPC_ACC_UP_MAX
[ERROR] [1590423455.493307820]: PR: Unknown parameter to get: MPC_XY_CRUISE
[ERROR] [1590423455.495081955]: PR: Unknown parameter to get: MPC_Z_VEL_MAX_DN
[ERROR] [1590423455.496867093]: PR: Unknown parameter to get: MPC_Z_VEL_MAX_UP
[ERROR] [1590423455.498607248]: PR: Unknown parameter to get: CP_DIST
[ERROR] [1590423455.500313905]: PR: Unknown parameter to get: MPC_LAND_SPEED
[ERROR] [1590423455.501982579]: PR: Unknown parameter to get: MPC_JERK_MAX
[ERROR] [1590423455.503698172]: PR: Unknown parameter to get: NAV_ACC_RAD
[ERROR] [1590423455.505402975]: PR: Unknown parameter to get: MPC_YAWRAUTO_MAX
[ERROR] [1590423460.507398409]: PR: Unknown parameter to get: MPC_ACC_HOR
[ERROR] [1590423460.509238109]: PR: Unknown parameter to get: MPC_ACC_DOWN_MAX
[ERROR] [1590423460.511016848]: PR: Unknown parameter to get: MPC_ACC_UP_MAX
[ERROR] [1590423460.512949144]: PR: Unknown parameter to get: MPC_XY_CRUISE
[ERROR] [1590423460.514785767]: PR: Unknown parameter to get: MPC_Z_VEL_MAX_DN
[ERROR] [1590423460.516488228]: PR: Unknown parameter to get: MPC_Z_VEL_MAX_UP
[ERROR] [1590423460.518188251]: PR: Unknown parameter to get: CP_DIST
[ERROR] [1590423460.520177168]: PR: Unknown parameter to get: MPC_LAND_SPEED
[ERROR] [1590423460.521881722]: PR: Unknown parameter to get: MPC_JERK_MAX
[ERROR] [1590423460.523602578]: PR: Unknown parameter to get: NAV_ACC_RAD
[ERROR] [1590423460.525297964]: PR: Unknown parameter to get: MPC_YAWRAUTO_MAX

```

codevegan, 2020-06-06:
I also get the same error information. @mrivi  Could you help to see the issue?

codevegan, 2020-06-06:
It takes a long time to start, it can work normally after waiting for a while

Cristian-wp, 2020-06-08:
@codevegan have you setup a parameter? I remember that I had to set a parameter to TRUE in order to start it, but I do not remember which one...
It was enable simulation or something like that... @mrivi can you confirm this or was just a case?

codevegan, 2020-06-10:
No, I don't need to setup the parameter

Jaeyoung-Lim, 2020-06-10:
@codevegan The reason it takes long to startup is not coming from this repo, but the fact that gazebo is trying to download the tree models from the model database

001zwzwzw, 2022-03-24:
@Jaeyoung-Lim i also meet the question(unknow parameter to get) ,can you tell me how to solve it ?thanks

ritvikmahajan01, 2022-08-30:
I'm also facing the same errors (On Ubuntu 20.04). Has anyone found a fix yet?

2447643603jzp, 2022-09-11:
@ritvikmahajan01  maybe you should change the file named "avoidance_sitl_mavros.launch".you can change the "gui" of "false"to"true"。then you can see when the gazebo is opened.when it is opened ,you can get the parameter and the error will disappear. the only thing is that you should wait a little time until the gazebo is opened.  may be my english is not good ,sorry.

cr5519, 2023-02-02:
I am having this issue even though the simulation is fully loaded.

ChanJoon, 2023-05-13:
I faced the same errors on Ubuntu 20.04, but after following the instructions correctly, I modified the "avoidance_sitl_mavros.launch" file as suggested by user @2447643603jzp (changing the argument "gui" from false to true).

It may take a long time, depending on your hardware specifications. However, after waiting, Gazebo finally opened, and after some additional waiting, the issue of "PR: Unknown parameter to get" stopped appearing.
This resolved the problem.

Vu1tUrR, 2023-09-11:
> I am having this issue even though the simulation is fully loaded.

Gazebo is downloading models, so please wait, it depends upon your internet speed, once it is done, the simulation will stat