PX4/PX4-Avoidance #584: FCU  avoidance system connecting and time out over and over

AldoCF96, 2020-05-27:
these are the outputs of running:
`roslaunch local_planner avoidance.launch fcu_url:=/dev/ttyACM0:57600`
`rostopic hz /local_pointcloud`

[image]

the avoidance system is connecting and disconnecting, I am not sure what is the error ?

baumanta, 2020-05-27:
Your pointcloud rate is way too low. We recommend using something around 10-15Hz. The planner complains (timeout) because it does not get the data at the right frequency

AldoCF96, 2020-05-27:
I havent changed the pointcloud rate. I just took the files as they were, where do I change the pointcloud rate ?

AldoCF96, 2020-05-27:
I tried changing the fps but it did not help

baumanta, 2020-05-27:
most likely it is a hardware limitation of your laptop. The code as prided works, you should not need to change anything (e.g. frame rates) for SITL. Try running it on a more powerful computer, maybe a desktop.

AldoCF96, 2020-05-27:
Yeah, I will try that, as of now I am running in actual hardware.  so I am not using SITL right now,  any recommended hardware ?

baumanta, 2020-05-28:
So we successfully had it running on: Nvidia jetson TX2/TX1, Nvidia IMX8, Intel Up Core, Intel NUC
With any of these or comparable boards you would be on the safe side

baumanta, 2020-05-28:
Closing as this is no code issue

closed 2020-05-28 by baumanta

AldoCF96, 2020-05-28:
what is the intel UP core rate?? I was working with a board of similar specs and got around 9.6 hz is it good ?

baumanta, 2020-05-29:
That should do. And you will for sure not see the timeout anymore