PX4/PX4-Avoidance #587: Set Goal in Gazebo

Sebboo97, 2020-05-29:
I'm currently evaluating the _local_planner_ with _depth camera_ in simulation.
For that I have imported my own _.stl_ file of an Industry hall as a gazebo model and saved it as a new world, see below.
At the moment I open Rviz and Gazebo at the same time because I need Rviz for publishing goals and Gazebo for showing my created world and checking if the drone crashes.
The problem is, that Rviz does not show my Industry hall (Gazebo model) so the clicked goal is pretty random and I can't check if it's actually free.

[image]

Is there a way to do all that in **one program**, like setting goals in Gazebo and display the drone's path?
Or maybe display my own world with Industry hall in Rviz?

baumanta, 2020-06-02:
If your world consists of simple shapes (which I think it does), you can create a .yaml file withe same name as your .world file. look at the other files in avoidance/sim/worlds as an example. The yaml file will be imported in RVIZ and show your world

pkyadav73199, 2020-06-05:
@baumanta  can you please suggest me.
how would i create yaml file of test_city_2.world file ??

baumanta, 2020-06-08:
@pkyadav73199 Look at the other yaml file in this repo and take them as examples. I'm sure you can figure it out

closed 2020-06-08 by baumanta