PX4/PX4-Avoidance #588: TF Buffer: could not retrieve requested transform from buffer, unregistered. Drone not showing in rviz.

ChenChangHaoWS, 2020-06-04:
**Describe the bug**
TF Buffer: could not retrieve requested transform from buffer, unregistered. 
Drone not showing in rviz, but can be seen in gazebo.

**Checked by myself**
[image]
[image]
[frames.pdf](https://github.com/PX4/avoidance/files/4730758/frames.pdf)

**Log**
[image]

ChenChangHaoWS, 2020-06-04:
**Full log is here:**
... logging to /home/nvidia/.ros/log/3ef27362-a670-11ea-96fe-00044bded595/roslaunch-nvidia-desktop-21101.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nvidia-desktop:34097/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /local_planner_node/goal_x_param: 15
 * /local_planner_node/goal_y_param: 15
 * /local_planner_node/goal_z_param: 4
 * /local_planner_node/pointcloud_topics: ['/stereo/points2']
 * /local_planner_node/world_name: /home/nvidia/work...
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://:14540@loca...
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.0003490659 // 0...
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: local_origin
 * /mavros/local_position/tf/child_frame_id: fcu
 * /mavros/local_position/tf/frame_id: local_origin
 * /mavros/local_position/tf/send: True
 * /mavros/local_position/tf/send_fcu: False
 * /mavros/mission/pull_after_gcs: True
 * /mavros/mocap/use_pose: False
 * /mavros/mocap/use_tf: True
[... 511 lines cut ...]
[ERROR] [1591281179.574539720, 4.044000000]: PR: Unknown parameter to get: MPC_ACC_UP_MAX
[ERROR] [1591281179.578857936, 4.044000000]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1591281179.579563217, 4.044000000]: PR: Unknown parameter to get: MPC_XY_CRUISE
[ERROR] [1591281179.584727515, 4.044000000]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1591281179.585747325, 4.044000000]: PR: Unknown parameter to get: MPC_Z_VEL_MAX_DN
[ERROR] [1591281179.595209262, 4.048000000]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1591281179.596792913, 4.052000000]: PR: Unknown parameter to get: MPC_Z_VEL_MAX_UP
[ERROR] [1591281179.600306712, 4.052000000]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1591281179.602782972, 4.052000000]: PR: Unknown parameter to get: CP_DIST
[ERROR] [1591281179.606555395, 4.056000000]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1591281179.610803403, 4.064000000]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1591281179.612767535, 4.064000000]: PR: Unknown parameter to get: MPC_LAND_SPEED
[ERROR] [1591281179.616109909, 4.064000000]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1591281179.622181760, 4.068000000]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1591281179.626534280, 4.076000000]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1591281179.631640466, 4.080000000]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1591281179.636739675, 4.080000000]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1591281179.637363037, 4.084000000]: PR: Unknown parameter to get: MPC_JERK_MAX
[ERROR] [1591281179.641859077, 4.088000000]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1591281179.642534470, 4.088000000]: PR: Unknown parameter to get: NAV_ACC_RAD
[ERROR] [1591281179.646914094, 4.092000000]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1591281179.648246321, 4.092000000]: PR: Unknown parameter to get: MPC_YAWRAUTO_MAX
[ERROR] [1591281179.652026232, 4.096000000]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1591281179.657137761, 4.100000000]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1591281179.662228747, 4.108000000]: TF Buffer: could not retrieve requested transform from buffer, unregistered

DengLiankang, 2020-08-31:
In my opinion, this should be a problem of model import.

My solution is as follows：

- ensure that the GAZEBO_MODEL_PATH has been set correctly.

- delete all files under ~/.gazebo/models.

I think there may be a conflict between models.

hope this helps, If it doesn’t work, I don’t know other ways……

yxt916, 2020-09-06:
solution not working
same error and another error in rviz :
Global state error
Fixed Frame [local_origin] does not exist

notoriou5, 2021-02-26:
Hi @yxt916,
did you find any solution to your problem? I am having the same :(

LLeeSG, 2021-07-20:
> In my opinion, this should be a problem of model import.
>
> My solution is as follows：
>
> * ensure that the GAZEBO_MODEL_PATH has been set correctly.
> * delete all files under ~/.gazebo/models.
>
> I think [...]

pity that it didn't work, anyone got the solution?

Cristian-wp, 2022-11-24:
Same error, do you have a solution? I am on Ubuntu 18.04 and running firmware v1.13.2 (same problem with 1.11.3)