PX4/PX4-Avoidance #589: ERROR : can't launch node of type [local_planner/local_planner_node]

sammar80, 2020-06-23:
Hello,
after I followed the installation instructions I get this error  when I run 
roslaunch local_planner local_planner_stereo.launch 
(ERROR : can't launch node of type [local_planner/local_planner_node] : Can't locate node of type [local_planner_node] in package [local_planner].Make sure the file exists in the package path and permission is set to executable )

I found that the local_planner_node.h and the local_planner_node.cpp are missing from the master stream  . Please advice

baumanta, 2020-06-23:
try sourcing: `source /catkin_ws/devel/setup.bash`

sammar80, 2020-06-23:
Thank you baumanta , 
I did that but it didn't work , I found the files in another branch