PX4/PX4-Avoidance #592: Goalpoint misscommunication

Mekiso, 2020-07-08:
**Describe the bug**
I need to run the local planner in offboard mode. When I start the local planner, the goal point is not correctly set. X and Y coordinates are correct, however the z coordinate is 0. If I change the rosparam /local_planner_nodelet/goal_x_param (and y and z), this is not published to rviz (or received by rviz). Manually changing the 2D Goal point in rviz doesn't have an effect on the rosparams either. Setting goal_z_param in rqt_config, however, changes the rosparam in the terminal. Setting goal_z_param via the terminal does not change the goal position in rvis and rqt_config.
In previous releases (0.3.0.) it was defenetly possible to set goal points by adding them in the launch file as well (e.g. by writing this: "/launch/node//param[@name='goal_x_param']/@value" -v "$goal_x" into the launchfile with xmlstarlet).  Were there any changes made to the subscriber nodes? 

_Do you have any suggestions on how to set the goal point in OFFBOARD mode via the terminal with the current (0.3.1 release)_

**To Reproduce**
e.g. roslaunch local_planner local_planner_sitl_3cam.launch (same for other launch files)
the goal position in rviz is at zero height.
[image]

additionally:
please compare the goal_z_param in the terminal and in rqt_config in the picture below.
[image]

I changed it manually in rqt_reconf, the yellow ball moves, I changed it in the terminal, no influence on rviz.

**Expected behavior**
Change of the rviz and locla path planner goal position according to the rosparam goal_x/y/z_param

**log**
%log - Note: My Computer is quite slow. I suppose the errors are generated by a subscriber which tries to receive parameters which are not published yet because it's still starting. Additionally, I get a warning when switching to OFFBOARD, which does not seem to be an actuall issue (see for example here:  https://github.com/mavlink/mavros/issues/620 or here https://github.com/mavlink/mavros/issues/536)

roslaunch local_planner local_planner_sitl_3cam.launch ... logging to /home/linux123/.ros/log/cf5ee682-c0f7-11ea-8b0e-0023ae90f082/roslaunch-HiWi-Rechner-3449.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://HiWi-Rechner:36055/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /local_planner_nodelet/goal_x_param: 20
 * /local_planner_nodelet/goal_y_param: 10
 * /local_planner_nodelet/goal_z_param: 3
 * /local_planner_nodelet/pointcloud_topics: ['/camera_front/d...
 * /local_planner_nodelet/world_name: /home/linux123/ca...
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
[... 306 lines cut ...]

[ INFO] [1594198136.821489432, 11.524000000]: parameter MPC_XY_CRUISE is set from  nan to 5.000000 

[ INFO] [1594198136.830601626, 11.532000000]: parameter MPC_YAWRAUTO_MAX is set from  nan to 45.000000 

[ INFO] [1594198141.633765370, 16.252000000]: WP: mission received
[ WARN] [1594198160.190378895, 33.476000000]: CMD: Unexpected command 176, result 0
WARN  [mc_pos_control] Offboard activation failed with error: Activation Failed
WARN  [mc_pos_control] Position-Ctrl activation failed with error: Activation Failed
WARN  [mc_pos_control] Altitude-Ctrl activation failed with error: Activation Failed
INFO  [commander] ARMED by Arm/Disarm component command
[ INFO] [1594198168.734024899, 41.544000000]: FCU: ARMED by Arm/Disarm component command
INFO  [commander] DISARMED by Auto disarm initiated
[ INFO] [1594198179.612510336, 51.568000000]: FCU: DISARMED by Auto disarm initiated
INFO  [logger] closed logfile, bytes written: 2840633
[ INFO] [1594198201.502420280, 71.840000000]: Setting goal: Frame:local_origin, Position(10,713, -11,919, 0,000), Orientation(0,000, 0,000, -0,292, 0,956) = Angle: -0,593

INFO  [commander] ARMED by Arm/Disarm component command
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2020-07-08/08_50_01.ulg
INFO  [logger] Opened full log file: ./log/2020-07-08/08_50_01.ulg
[ INFO] [1594198208.326838091, 78.356000000]: FCU: ARMED by Arm/Disarm component command
[ INFO] [1594198208.383548477, 78.408000000]: FCU: [logger] ./log/2020-07-08/08_50_01.ulg
INFO  [commander] Takeoff detected
[ INFO] [1594198209.526229716, 79.492000000]: FCU: Takeoff detected

Mekiso, 2020-07-24:
I found a mistake in the launch file loading path of the goal_z_param, unfortunately it is still not fixing the issue:

https://github.com/PX4/avoidance/commit/0e61dc91fee12c8743743783189e58ca20c87f8b#r40874226

Seems like the method LocalPlannerNode::readParams() is never called correctly? 

Whereelse in the code is the local planner loading the x,y parameter, but not the z parameter?

Adidushi, 2021-03-02:
Hi, has this been updated? I can't change the Z param without dynamic reconfigure at the moment.

Adidushi, 2021-03-03:
I've solved this issue by changing the code to allow setting the z coordinate through the same pose as the x and y coordinates:  
Commented out [line 328 in `local_planner/src/nodes/local_planner_nodelet.cpp`](https://github.com/PX4/PX4-Avoidance/blob/41d057552ba6516dca81ea327044f96d9aeab53d/local_planner/src/nodes/local_planner_nodelet.cpp#L328)

RapIt, 2021-06-18:
I guess the reason is a typo in https://github.com/PX4/PX4-Avoidance/blob/dafe72aabfdefc2dc409e28f7318717194f1bfa6/local_planner/src/nodes/local_planner_nodelet.cpp#L124 . It tries to read the rosparam `lgoal_z_param` instead of `goal_z_param`.