PX4/PX4-Avoidance #593: Obstacle Avoidance for Drones Using a 3DVFH* Algorithm

zhanqx1024, 2020-07-09:
I see the local planner avoidance is based on the 3DVFH+,but It's explaintion is too simple , I find you have reference the paper 3DVFH*, where can I get the paper of Obstacle Avoidance for Drones Using a 3DVFH* Algorithm, I can not find it in the  EI and Reaserchgate .Thank you very much

nicovanduijn, 2020-07-13:
It's not on researchgate. This is the only paper on it: https://drive.google.com/file/d/1yjDtxRrIntr5Mdaj9CCB4IFJn0Iy2-bR/view

closed 2020-07-13 by nicovanduijn

ho4ngt, 2023-02-23:
> It's not on researchgate. This is the only paper on it: https://drive.google.com/file/d/1yjDtxRrIntr5Mdaj9CCB4IFJn0Iy2-bR/view

Can you please share this paper again? That link is no longer working. Thank you!

nicovanduijn, 2023-02-24:
It's been updated on the readme a while ago with #690

ho4ngt, 2023-02-24:
> It's been updated on the readme a while ago with #690

Thank you!