PX4/PX4-Avoidance #594: TF Buffer: could not retrieve requested transform from buffer, unregistered

Cristian-wp, 2020-07-16:
### Avoidance version and platform

Avoidance: 0.3.1
ROS: Melodic
Ubuntu: 18.04 LTS

### MAVROS version and platform

Mavros: 1.2.0
ROS: Melodic
Ubuntu: 18.04 LTS

### Autopilot type and version

[  ] ArduPilot
[ x ] PX4

Version: 1.10.1 stable

### Issue details

Good morning,
I am trying to test the collision avoidance package (local planner) on real hardware.
I have follow the instruction reported on the [repository](https://github.com/PX4/avoidance), but I can not manage to make it work.

As camera I use a Intel Realsence D435i, already tested in RViz and it works.

I have generate the launch file using the instruction provided "./tools/generate_launchfile.sh" with my parameters, but when I start the test I get this output:

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /D435i/realsense2_camera/accel_fps: 0
 * /D435i/realsense2_camera/accel_frame_id: D435i_accel_frame
 * /D435i/realsense2_camera/accel_optical_frame_id: D435i_accel_optic...
 * /D435i/realsense2_camera/align_depth: False
 * /D435i/realsense2_camera/aligned_depth_to_color_frame_id: D435i_aligned_dep...
 * /D435i/realsense2_camera/aligned_depth_to_fisheye1_frame_id: D435i_aligned_dep...
 * /D435i/realsense2_camera/aligned_depth_to_fisheye2_frame_id: D435i_aligned_dep...
 * /D435i/realsense2_camera/aligned_depth_to_fisheye_frame_id: D435i_aligned_dep...
 * /D435i/realsense2_camera/aligned_depth_to_infra1_frame_id: D435i_aligned_dep...
 * /D435i/realsense2_camera/aligned_depth_to_infra2_frame_id: D435i_aligned_dep...
 * /D435i/realsense2_camera/allow_no_texture_points: False
 * /D435i/realsense2_camera/base_frame_id: D435i_link
 * /D435i/realsense2_camera/calib_odom_file: 
 * /D435i/realsense2_camera/clip_distance: -1.0
 * /D435i/realsense2_camera/color_fps: 15
 * /D435i/realsense2_camera/color_frame_id: D435i_color_frame
 * /D435i/realsense2_camera/color_height: 480
 * /D435i/realsense2_camera/color_optical_frame_id: D435i_color_optic...
 * /D435i/realsense2_camera/color_width: 640
 * /D435i/realsense2_camera/depth_fps: 30
 * /D435i/realsense2_camera/depth_frame_id: D435i_depth_frame
 * /D435i/realsense2_camera/depth_height: 480
 * /D435i/realsense2_camera/depth_optical_frame_id: D435i_depth_optic...
 * /D435i/realsense2_camera/depth_width: 640
 * /D435i/realsense2_camera/device_type: 
 * /D435i/realsense2_camera/enable_accel: False
 * /D435i/realsense2_camera/enable_color: False
 * /D435i/realsense2_camera/enable_depth: True
 * /D435i/realsense2_camera/enable_fisheye1: False
 * /D435i/realsense2_camera/enable_fisheye2: False
 * /D435i/realsense2_camera/enable_fisheye: False
 * /D435i/realsense2_camera/enable_gyro: False
 * /D435i/realsense2_camera/enable_infra1: True
 * /D435i/realsense2_camera/enable_infra2: False
[... 413 lines cut ...]
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument

I have already take a look on similar issues:

#589 
#588 
#523 
#578 
#547 this one is exactly my problem, but I do not understand what I have do to to fix it.

I think to understand that the problem is inside px4_config.yaml, I have try to fix it but with any results...

Using RVIZ I can see the pointcloud choosing as Fixed Frame D435i_depth_frame and publish the topic /D435i/depth/points.

Using rqt_graph I optain this result:

[image]

I see that the TF node of the camera is connected to nothing, and the normal TF node generated by mavros is not present so I think this is the problem, but I do not have a clue how to fix it.

Can someone help me please?

closed 2020-07-20 by Cristian-wp

Abner0907, 2020-08-28:
have you fix this problem ? I have the same as you.

Cristian-wp, 2020-08-31:
No :(

Abner0907, 2020-08-31:
@Cristian-wp  I have your similar problem. But now I fix it.
I have seen your post above. And add your configuration below, you can post and have a try.

~~~
export CAMERA_CONFIGS="D435i,realsense,021122070141,0.3,0.32,-0.11,0,0,0"
export DEPTH_CAMERA_FRAME_RATE=30
export VEHICLE_CONFIG=~/catkin_ws/src/avoidance/local_planner/cfg/vehicle.yaml
./tools/generate_launchfile.sh

LLeeSG, 2021-07-20:
> export CAMERA_CONFIGS="D435i,realsense,021122070141,0.3,0.32,-0.11,0,0,0"
> export DEPTH_CAMERA_FRAME_RATE=30
> export VEHICLE_CONFIG=~/catkin_ws/src/avoidance/local_planner/cfg/vehicle.yaml
> ./tools/generate_launchfile.sh

still didn't work, anyone knows other solution?

ChanJoon, 2023-04-15:
@Cristian-wp @Abner0907 @lleesg HI guys, I still haven't solved the problem, I was wondering if any of you have solved it and could give me some advice?