PX4/PX4-Avoidance #596: Resource not found: px4 px4/Obstacle Avoidance Issue

JohnMarvinNeuhaus, 2020-07-26:
Hey guys
I hope i ll find some help here. So I followed the instructions on https://github.com/PX4/avoidance#obstacle-detection-and-avoidance. I installed everything beside Mavros. I Installed mavros just from the command sudo apt install ros-melodic-mavros ros-melodic-mavros-extras. When I install it from source the error is just bigger but its still the same error.

At the end when I want to launch roslaunch local_planner local_planner_stereo.launch or any other roslaunch I get this message: 

Resource not found: px4
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/neuhaus/catkin_ws/src/avoidance/avoidance
ROS path [2]=/home/neuhaus/catkin_ws/src/avoidance/global_planner
ROS path [3]=/home/neuhaus/catkin_ws/src/avoidance/local_planner
ROS path [4]=/home/neuhaus/catkin_ws/src/avoidance/safe_landing_planner
ROS path [5]=/opt/ros/melodic/share
The traceback for the exception was written to the log file

ROS Package Path: ROS_PACKAGE_PATH

GAZEBO_PLUGIN_PATH :/home/neuhaus/Firmware/build/px4_sitl_default/build_gazebo

GAZEBO_MODEL_PATH :/home/neuhaus/Firmware/Tools/sitl_gazebo/models

LD_LIBRARY_PATH /home/neuhaus/catkin_ws/devel/lib:/opt/ros/melodic/lib:/home/neuhaus/Firmware/build/px4_sitl_default/build_gazebo

The ROS Package Path is obviously not correct. But it doesnt matter what I do the Path doesnt change.

Can anyone help me please?

bwassim, 2020-08-10:
I am having exactly the same issue. if anyone has a clue, that would be great thanks

mohamedsayed18, 2020-11-30:
I was facing a similar issue. the problem was about the environment variables. I set them in the `.bashrc` with these values and it worked fine
```
px4_dir=~/Firmware
source /opt/ros/noetic/setup.bash
source ~/drone_ws/devel/setup.bash
source $px4_dir/Tools/setup_gazebo.bash $px4_dir $px4_dir/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$px4_dir
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$px4_dir/Tools/sitl_gazebo
export GAZEBO_PLUGIN_PATH=$GAZEBO_PLUGIN_PATH:/usr/lib/x86_64-linux-gnu/gazebo-11/plugins
@JohnMarvinNeuhaus I hope you had solved your problem by the time

closed 2021-03-23 by Jaeyoung-Lim

ahmadshobake, 2022-06-06:
cd Firmware
then run full as below :
. ~/Firmware/Tools/setup_gazebo.bash ~/Firmware ~/Firmware/build/px4_sitl_default && export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:~/Firmware && echo "export GAZEBO_MODEL_PATH=${GAZEBO_MODEL_PATH}:~/catkin_ws/src/avoidance/avoidance/sim/models:~/catkin_ws/src/avoidance/avoidance/sim/worlds" >> ~/.bashrc && roslaunch local_planner local_planner_stereo.launch