PX4/PX4-Avoidance #600: error while using roslaunch local_planner local_planner_stereo.launch

bwassim, 2020-08-10:
After running Gazebo and the following command
**make px4_sitl_default gazebo**
**roslaunch local_planner local_planner_stereo.launch**
I am having the following error below. The small part below is a part of the full output I got
                               

-----------------------------------------------------

process[stereo/stereo_image_proc-8]: started with pid [6424]
/opt/ros/melodic/lib/stereo_image_proc/stereo_image_proc: error while loading shared libraries: libwebp.so.5: cannot open shared object file: No such file or directory
process[local_planner_node-9]: started with pid [6425]
process[rviz-10]: started with pid [6426]
[sitl_0-4] process has died [pid 6411, exit code 255, cmd /home/obara/PX4/Firmware/build/px4_sitl_default/bin/px4 /home/obara/PX4/Firmware/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -i 0 __name:=sitl_0 __log:=/home/obara/.ros/log/5be7266c-db3f-11ea-a009-c8d9d20a312d/sitl_0-4.log].
log file: /home/obara/.ros/log/5be7266c-db3f-11ea-a009-c8d9d20a312d/sitl_0-4*.log
[stereo/stereo_image_proc-8] process has died [pid 6424, exit code 127, cmd /opt/ros/melodic/lib/stereo_image_proc/stereo_image_proc __name:=stereo_image_proc __log:=/home/obara/.ros/log/5be7266c-db3f-11ea-a009-c8d9d20a312d/stereo-stereo_image_proc-8.log].
log file: /home/obara/.ros/log/5be7266c-db3f-11ea-a009-c8d9d20a312d/stereo-stereo_image_proc-8*.log

-----------------------------------------------------------------------------------

This seemed to generate another error, not sure if both errors are correlated or not:

[gazebo-6] process has died [pid 6413, exit code 255, cmd /opt/ros/melodic/lib/gazebo_ros/gzserver -e ode /home/obara/catkin_ws/src/avoidance/avoidance/sim/worlds/simple_obstacle.world __name:=gazebo __log:=/home/obara/.ros/log/5be7266c-db3f-11ea-a009-c8d9d20a312d/gazebo-6.log].
log file: /home/obara/.ros/log/5be7266c-db3f-11ea-a009-c8d9d20a312d/gazebo-6*.log

------------------
The extracted above section can be found in the full output below 
Thank you for your help.

Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server .....

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /local_planner_node/goal_x_param: 15
 * /local_planner_node/goal_y_param: 15
 * /local_planner_node/goal_z_param: 4
 * /local_planner_node/pointcloud_topics: ['/stereo/points2']
 * /local_planner_node/world_name: /home/obara/catki...
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
[... 374 lines cut ...]
[ERROR] [1597087617.129254413]: PR: Unknown parameter to get: MPC_JERK_MAX
[ERROR] [1597087617.132390241]: PR: Unknown parameter to get: NAV_ACC_RAD
[ERROR] [1597087617.135614575]: PR: Unknown parameter to get: MPC_YAWRAUTO_MAX

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Drone switched on '...'
2. Uploaded mission '....' (attach QGC mission file)
3. Took off '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/en/getting_started/flight_reporting.html)) or from the Gazebo simulation (logs are located in `~/.ros/logs/`)
- Share the link to a log showing the problem on [PX4 Flight Review](http://logs.px4.io/).
- Share a link to a rosbag showing the problem

**Additional context**
Add any other context about the problem here.

Abner0907, 2020-08-28:
Have you fix this problem? I have the same issues.

yxt916, 2020-09-04:
sitl0_4+gazebo6 errors and many PR: Unknown parameter to get errors

Phillweston, 2021-03-05:
Me too! Waiting to fix.

KingStrong-bit, 2021-03-15:
I meet the same problem. Did anyone solve this problem?

ataparlar, 2021-04-08:
Are you sure that you have installed the **stereo_image_proc** package?

Try:
`sudo apt install ros-melodic-stereo-image-proc`

You should source the workspace you using. I recommend to read the instructions in the [PX4-Avoidance](https://github.com/PX4/PX4-Avoidance#local-planner-default-heavily-flight-tested) repo.