PX4/PX4-Avoidance #605: yaml file not applied

lgh5054, 2020-09-18:
**Describe the bug**

We are testing with the px4 vision kit 

The structure.yaml file, which is set by default, does not apply.

**Expected behavior**
By default, the Yaml file is not applied, so only the parameters written in the local_planner_node changed the value to create a new Yaml file

The Yaml produced only by the parameters on the local_planner node is well applied.

However, the local_planner node does not have a parameter that adjusts the box_size, so you cannot change it if you want to.

I think it only applies when creating a Yaml file with the parameters listed in the Local_planner_node. How do I change the Box_size?

Jaeyoung-Lim, 2020-09-23:
@lgh5054 Which yaml file are you talking about?

lgh5054, 2020-09-23:
@Jaeyoung-Lim !!python/object/new:dynamic_reconfigure.encoding.Config
dictitems:
  adapt_cost_params_: true
  box_radius_: 15.0
  children_per_node_: 50
  goal_cost_param_: 10.0
  goal_z_param: 3.0
  groups: !!python/object/new:dynamic_reconfigure.encoding.Config
    dictitems:
      adapt_cost_params_: true
      box_radius_: 15.0
      children_per_node_: 50
      goal_cost_param_: 10.0
      goal_z_param: 3.0
      groups: !!python/object/new:dynamic_reconfigure.encoding.Config
        state: []
      heading_cost_param_: 0.5
      id: 0
      max_path_length_: 3.0
      max_point_age_s_: 20.0
      min_num_points_per_cell_: 100
      min_realsense_dist_: 0.2
      n_expanded_nodes_: 10
      name: Default
      no_progress_slope_: -0.0007
      parameters: !!python/object/new:dynamic_reconfigure.encoding.Config
        state: []
      parent: 0
      smooth_cost_param_: 1.5
      smoothing_margin_degrees_: 40.0
      smoothing_speed_xy_: 10.0
      smoothing_speed_z_: 3.0
      state: true
      timeout_critical_: 0.5
      timeout_termination_: 15.0
      tree_discount_factor_: 0.8
      tree_node_distance_: 1.0
      type: ''
      use_vel_setpoints_: false
    state: []
  heading_cost_param_: 0.5
  max_path_length_: 3.0
  max_point_age_s_: 20.0
  min_num_points_per_cell_: 100
  min_realsense_dist_: 0.2
  n_expanded_nodes_: 10
  no_progress_slope_: -0.0007
  smooth_cost_param_: 1.5
  smoothing_margin_degrees_: 40.0
  smoothing_speed_xy_: 10.0
  smoothing_speed_z_: 3.0
  timeout_critical_: 0.5
  timeout_termination_: 15.0
  tree_discount_factor_: 0.8
  tree_node_distance_: 1.0
  use_vel_setpoints_: false
state: []

Jaeyoung-Lim, 2020-09-23:
@lgh5054 and...where is this yaml file?

Jaeyoung-Lim, 2020-09-23:
@lgh5054 I am not sure if I understand correctly, is this a question about this repository or the px4vision?

lgh5054, 2020-09-24:
@Jaeyoung-Lim Both of these questions apply.

The parametas above are yaml applied to the px4 vision kit. However, the local_planner used in the px4vision kit was the latest version, so the box_size and other Parameta were changed to the latest version and disappeared. Therefore, the Yaml enclosed in the vision kit does not seem to be applicable.