PX4/PX4-Avoidance #606: How to change world

pirate-lofy, 2020-10-16:
I tried to change the world that starts with the simulator..
I sent the world_file_name argument as a parameter but it did not work
I modified the launch files on global_planner like global_planner_stereo.launch and changed the world_path argument to the path of another world in "avoidance/sim/worlds" but did not work..
I did the same modifications to the "avoidance/launch/avoidance_sitl_stereo.launch" and "avoidance/launch/avoidance_sitl_mavros.launch" but did not achieve anything
the whole global_planner_stereo.launch is 

```
<launch>
    <arg name="world_file_name"    default="simple_obstacle" />
    <arg name="world_path" default="~/catkin_ws/src/avoidance/avoidance/sim/worlds/outdoor_village_3.world" />
    <arg name="pointcloud_topics" default="[/stereo/points2]" />
    <arg name="start_pos_x" default="0.5" />
    <arg name="start_pos_y" default="0.5" />
    <arg name="start_pos_z" default="3.5" />

  <!-- Launch PX4 and mavros -->
  <include file="$(find avoidance)/launch/avoidance_sitl_stereo.launch" >
    <arg name="model" value="iris_stereo_camera" />
    <arg name="world_path" value="$(arg world_path)" />
    <arg name="pointcloud_topics" value="$(arg pointcloud_topics)"/>
  </include>

    <!-- Global planner -->
    <include file="$(find global_planner)/launch/global_planner_octomap.launch" >
        <arg name="start_pos_x" value="$(arg start_pos_x)" />
        <arg name="start_pos_y" value="$(arg start_pos_y)" />
        <arg name="start_pos_z" value="$(arg start_pos_z)" />
        <arg name="pointcloud_topics" value="$(arg pointcloud_topics)"/>
    </include>

    <!-- RViz -->
    <node pkg="rviz" type="rviz" output="screen" name="rviz" respawn="true"
          args="-d $(find global_planner)/resource/global_planner.rviz" />
</launch>
```

pirate-lofy, 2020-10-16:
UPDATE:
after many trials, I found that the world changed successfully but the rendering screen of rviz still shows the default world
[image]
that makes sense as the global_planner was moving very bad according to the default world..
How can I force rviz to render the current world

heleidsn, 2020-10-18:
Hi, you should also change the **world_file_name**.

closed 2021-04-30 by pirate-lofy