PX4/PX4-Avoidance #609: Obstacle avoidance error with a QAV 250 drone, a Jetson Nano and a PX4

AdonaiVera, 2020-10-28:
Hello Community.
We are implementing the obstacle avoidance system in local planner inside a Jetson Nano connected by telemetry to a px4 by Baud of 921600, the vehicle that we configured is a Holybro QAV250 with the firmware v1.10.1, and the configuration parameters for the Pixhawek mini from QgroundControl are:
- `COM_OBS_AVOID` = 1
- `MAV_1_CONFIG` =  TELEM 1
- `MAV_0_CONFIG` =  DISABLE
- `MAV_1_MODE` =  Onboard
- `SER_TEL1_BAUD` = 921600

The .launch we use was [.launch](https://github.com/PX4/PX4-Avoidance/blob/master/local_planner/launch/local_planner_aero.launch) and we create a node that publishes the setpoint in the theme: /move_base_simple/goal in OFFBOARD mode. In hitl simulation the system works perfectly with Gazebo and Rviz, however when we test it on the jetson nano, the system performs the take off and then starts to move without a fixed direction. If we check the logs we found that the GPS does not work properly, however in other tests in OFFBOARD (without the evasion module) works correctly:

Console Output:
WARN  [commander] Preflight: GPS Horizontal Pos Error too high
WARN  [commander] Arming denied! Press safety switch first
WARN  [commander] Preflight: GPS Horizontal Pos Error too high
WARN  [commander] Arming denied! Press safety switch first
WARN  [commander] Preflight Fail: horizontal velocity estimate not stable
WARN  [commander] Preflight: GPS Horizontal Pos Error too high
WARN  [commander] Arming denied! Press safety switch first
WARN  [commander] Preflight: GPS Horizontal Pos Error too high
WARN  [commander] Arming denied! Press safety switch first
WARN  [load_mon] wq:I2C1 low on stack! (292 bytes left)
WARN  [load_mon] wq:I2C1 low on stack! (292 bytes left)
WARN  [commander] REJECT POSITION CONTROL
WARN  [commander] REJECT POSITION CONTROL
WARN  [commander] Preflight Fail: horizontal velocity estimate not stable
WARN  [commander] Preflight Fail: horizontal velocity estimate not stable
WARN  [commander] Preflight Fail: horizontal velocity estimate not stable
WARN  [commander] Preflight: GPS Horizontal Pos Error too high
WARN  [commander] Failsafe enabled: no local position
INFO  [logger] Start file log (type: full)
INFO  [logger] Opened full log file: /fs/microsd/log/2020-10-27/22_17_15.ulg
WARN  [mc_pos_control] Auto activation failed with error: Activation Failed
WARN  [load_mon] wq:I2C1 low on stack! (292 bytes left)
INFO  [logger] closed logfile, bytes written: 449443
WARN  [load_mon] wq:I2C1 low on stack! (292 bytes left)
WARN  [load_mon] wq:I2C1 low on stack! (292 bytes left)
WARN  [commander] REJECT POSITION CONTROL
WARN  [commander] REJECT POSITION CONTROL
WARN  [commander] REJECT POSITION CONTROL
WARN  [commander] REJECT POSITION CONTROL
WARN  [load_mon] wq:I2C1 low on stack! (292 bytes left)
WARN  [commander] REJECT POSITION CONTROL
WARN  [load_mon] wq:I2C1 low on stack! (292 bytes left)
WARN  [commander] REJECT POSITION CONTROL
WARN  [load_mon] wq:I2C1 low on stack! (292 bytes left)
WARN  [commander] REJECT POSITION CONTROL
WARN  [load_mon] wq:I2C1 low on stack! (292 bytes left)
WARN  [load_mon] wq:I2C1 low on stack! (292 bytes left)
WARN  [load_mon] wq:I2C1 low on stack! (292 bytes left)
WARN  [commander] REJECT POSITION CONTROL
WARN  [load_mon] wq:I2C1 low on stack! (292 bytes left)
WARN  [commander] REJECT POSITION CONTROL
WARN  [commander] REJECT POSITION CONTROL
WARN  [commander] Preflight: GPS Horizontal Pos Error too high

[Log complete](https://logs.px4.io/plot_app?log=c55e75af-5704-43d7-b68f-d4b2f96af283)

This is the trajectory of the drone:
[image]

We have tried different techniques but did not find the error. We don't know what else to do, someone could help us solve it. 
Thanks for taking the time to read the issue, regards.

nicovanduijn, 2020-10-29:
Your GPS accuracy is about 7m horizontally... There is no way you can fly in position control like that.

My first guess would be that the jetson emits too much and jams it - but the jamming indicator looks quite okay in the log you shared. I suggest to double-check your GPS mounting and figure out how to get decent postion controlled flight before moving into anything autonomous. If you get there, the next step is to try increasing the system load (CPU, GPU) on the jetson. More often than not the system load is what causes GPS interference. Only when you can get good position controlled flight under full load, start "closing the loop" and fly autonomously

AdonaiVera, 2020-11-28:
@nicovanduijn  first thank you for your answer. We did all the changes that you recommended, the problem was caused for the realsense r200 connected to a USB3 Jetson Nano, when we connect this the interference increases a lot and this make not fix the GPS and the barometer. That's why we decide to use the principle of Faraday Shield and try again. Do you have any suggestions for that? Thanks in advance.

jkflying, 2020-11-30:
@AdonaiVera I suggest you use a USB2 cable. We've done a lot of testing and unless you do serious RF engineering USB3 will cause significant GPS interference.

closed 2020-11-30 by jkflying

szobov, 2020-11-30:
@AdonaiVera I've made a small [article](https://blog.szobov.ru/2018/12/24/try-to-avoid-using-usb3.0-and-2.4hz-radio-like-gps/) some time ago, describing the same problem.
I've mentioned there [this](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/realsense-camera-r200-datasheet.pdf) manual, topic "shielding"