PX4/PX4-Avoidance #610: RLException: [global_planner_stereo.launch] is neither a launch file in package [global_planner] nor is [global_planner] a launch file name

pirate-lofy, 2020-11-08:
When I run this command 
`roslaunch global_planner global_planner_stereo.launch` 
it produces the following error
```
RLException: [global_planner_stereo.launch] is neither a launch file in package [global_planner] nor is [global_planner] a launch file name
The traceback for the exception was written to the log file
```
the same error occurs for global_planner , local_planner and safe_landing_planner (all avoidance package).

when I search for ros packages available I can not find any avoidance packages.. and when using 
`rospack find global_planner` 
the result be 
`[rospack] Error: package 'global_planner' not found`

Although all other ros packages located inside catkin_ws/src directory are available.
I reinstalled avoidance and Fimeware (and build catkin) many times but I had no luck.

MertEgeCAN, 2020-11-24:
Did you pull the repository into your `catkin_ws/src` and compiled it with `catkin_make` or `catkin build`? Then sourced the `catkin_ws/devel/setup.bash` in your terminal or in `.bashrc`?

pirate-lofy, 2020-11-24:
I used catkin build and sourced the setup.bash file

pirate-lofy, 2021-01-05:
Solved..
I found that when using catkin build it produces this kind of errors, so when using catkin_make these errors disappear.

closed 2021-01-05 by pirate-lofy

zh-pang, 2021-06-21:
I use catkin_make,it still exists.