PX4/PX4-Avoidance #613: How to simulate multiple UAVs?

Prashant446, 2020-12-03:
I am trying to simulate multiple UAVs and then I plan to modify or write my own code for planning. I am using the following launch file:
```
<?xml version="1.0"?>
<launch>
  <!-- MAVROS posix SITL environment launch script -->
  <!-- launches Gazebo environment and 2x: MAVROS, PX4 SITL, and spawns vehicle -->

  <arg name="gui" default="true"/>
  <arg name="world_file_name" default="simple_obstacle" />
  <arg name="world_path" default="$(find avoidance)/sim/worlds/$(arg world_file_name).world" />
  <!-- Launch Gazebo -->
  <include file="$(find gazebo_ros)/launch/empty_world.launch">
    <arg name="gui" value="$(arg gui)"/>
    <arg name="world_name" value="$(arg world_path)" />
  </include>
  <!-- UAV0 -->
  <group ns="uav0">
    <!-- MAVROS and vehicle configs -->
    <arg name="ID" value="0"/>
    <arg name="fcu_url" default="udp://:14540@localhost:14580"/>

    <include file="$(find offb)/launch/global_planner_stereo.launch">
      <arg name="ID" value="$(arg ID)"/>
    </include>
  </group>
</launch>
```
I have only modified avoidance_sitl_mavros.launch to launch gazebo beforehand. Octomap is not working and I think the reason is some topics are not mapped to the namespace:

```
/actual_path
/blocked_marker
/camera/rgb/image_raw
/clicked_point
/clock
/closest_point
/closest_point_array
/cloud_in
/current_setpoint
/deg60_point
/deg60_point_array
/diagnostics
/drone
/drone_array
/explored_cells
/extended_marker
/gazebo/link_states
/gazebo/model_states
/gazebo/parameter_descriptions
/gazebo/parameter_updates
/gazebo/set_link_state
/gazebo/set_model_state
/global_goal
/global_temp_goal
/global_temp_path
/goal_position
/local_pointcloud
/mavlink/from
/mavlink/gcs_ip
/mavlink/to
/mavros/actuator_control
/mavros/adsb/send
/mavros/adsb/vehicle
/mavros/altitude
/mavros/battery
/mavros/cam_imu_sync/cam_imu_stamp
/mavros/companion_process/status
/mavros/debug_value/debug
/mavros/debug_value/debug_vector
/mavros/debug_value/named_value_float
[... 232 lines cut ...]
/uav0/stereo/right/image_rect_color/compressed
/uav0/stereo/right/image_rect_color/compressed/parameter_descriptions
/uav0/stereo/right/image_rect_color/compressed/parameter_updates
/uav0/stereo/right/image_rect_color/compressedDepth
/uav0/stereo/right/image_rect_color/compressedDepth/parameter_descriptions
/uav0/stereo/right/image_rect_color/compressedDepth/parameter_updates
/uav0/stereo/right/image_rect_color/theora
/uav0/stereo/right/image_rect_color/theora/parameter_descriptions
/uav0/stereo/right/image_rect_color/theora/parameter_updates
/uav0/stereo/stereo_image_proc/parameter_descriptions
/uav0/stereo/stereo_image_proc/parameter_updates
/uav0/stereo/stereo_image_proc_debayer_left/parameter_descriptions
/uav0/stereo/stereo_image_proc_debayer_left/parameter_updates
/uav0/stereo/stereo_image_proc_debayer_right/parameter_descriptions
/uav0/stereo/stereo_image_proc_debayer_right/parameter_updates
/uav0/stereo/stereo_image_proc_rectify_color_left/parameter_descriptions
/uav0/stereo/stereo_image_proc_rectify_color_left/parameter_updates
/uav0/stereo/stereo_image_proc_rectify_color_right/parameter_descriptions
/uav0/stereo/stereo_image_proc_rectify_color_right/parameter_updates
/uav0/stereo/stereo_image_proc_rectify_mono_left/parameter_descriptions
/uav0/stereo/stereo_image_proc_rectify_mono_left/parameter_updates
/uav0/stereo/stereo_image_proc_rectify_mono_right/parameter_descriptions
/uav0/stereo/stereo_image_proc_rectify_mono_right/parameter_updates
/world
```

KimHyungSub, 2021-04-29:
@Prashant446 Could you share any update about simulating multiple UAVs?
Actually, I have the same problem.

001zwzwzw, 2022-03-24:
@Prashant446 Could you share any update about simulating multiple UAVs?
Actually, I have the same problem.thanks！

Cavalletta98, 2022-12-27:
How do you get namespace near camera topics?