PX4/PX4-Avoidance #615: PX4 Gazebo HITL - No Drone Appears When Launching Gazebo

jmalbert97, 2020-12-10:
I have followed all instructions here: https://dev.px4.io/master/en/simulation/hitl.html attempting to get a very simple HITL sim running. I have pixhawk 4, running on Ubuntu 18.04 with the latest github firmware. After following all of the steps, when executing the command: gazebo _Tools/sitl_gazebo/worlds/iris.world_ with PIxhawk 4 plugged in via USB, Gazebo is launched, but an empty world is shown. Also, the flight controller does not blink or make any noises. When attempting the command: ./Tools/jmavsim_run.sh -q -s -d /dev/ttyACM0 -b 921600 -r 250, the simulation runs fine and the drone can be moved with the RC or QGroundControl. Any ideas why the drone would not show up in a HITL simulation for Gazebo? After browsing some forums, i attempted this command:  GAZEBO_MODEL_PATH=${GAZEBO_MODEL_PATH}:/home/my_user/my_catkin_ws/src/, which did not seem to make a difference. Thank you for any guidance you can give. 

[image]

Eskenaut, 2021-03-29:
Hello, 
Did you solve this ? I'm having the same problem and could use some help ..

Jaeyoung-Lim, 2021-03-29:
I don't think PX4 avoidance with HITL is a supported usecase