PX4/PX4-Avoidance #618: can't not repeat the successful local_planner simulation.

darkasassin44, 2020-12-27:
I have been struggling to represent the SITL with local planner for a few weeks.
First, I'm a new to ROS.
I'd just followed this instruction, and did the SITL at the first time for all the three launch files, `local_planner_stereo`, `local_planner_depth_camera`, and `local_planner_sitl_3cam` with QGroundControl.
From the beginning, I put the variables below to the ~/catkin_ws/devel/setup.bash file.
```
. ~/Firmware/Tools/setup_gazebo.bash ~/Firmware ~/Firmware/build/px4_sitl_default
export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:~/Firmware
export GAZEBO_MODEL_PATH=${GAZEBO_MODEL_PATH}:~/catkin_ws/src/avoidance/avoidance/sim/models:~/catkin_ws/src/avoidance/avoidance/sim/worlds
```

https://user-images.githubusercontent.com/54735854/103171799-6614db00-4892-11eb-931f-8ecef90a6dd9.mp4

I have 3 ubuntu machines, and all of the three laptops are showing same problem.

So I chose to start from beginning, removed 'avoidance' package from /catkin_ws, and /Firmware from my /home directory.
But in the building gazebo simulation section, when I did `make px4_sitl_default gazebo`, I could see the multiple vehicles on gazebo as below.
[image]

lbd_song@lbd-gcs:~$ roslaunch local_planner local_planner_depth-camera.launch 
... logging to /home/lbd_song/.ros/log/21e44136-484d-11eb-84b3-e0d4e8ff40e7/roslaunch-lbd-gcs-12904.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:46773/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /local_planner_nodelet/goal_x_param: 17
 * /local_planner_nodelet/goal_y_param: 15
 * /local_planner_nodelet/goal_z_param: 3
 * /local_planner_nodelet/pointcloud_topics: ['/camera/depth/p...
 * /local_planner_nodelet/world_name: /home/lbd_song/ca...
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://:14540@loca...
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
[... 213 lines cut ...]
[ INFO] [1609078169.439016129]: Built-in MAVLink package version: 2020.12.12
[ INFO] [1609078169.439044690]: Known MAVLink dialects: common ardupilotmega ASLUAV all autoquad icarous matrixpilot paparazzi standard uAvionix ualberta
[ INFO] [1609078169.439060719]: MAVROS started. MY ID 1.240, TARGET ID 1.1
[gazebo-6] process has died [pid 12966, exit code 255, cmd /opt/ros/melodic/lib/gazebo_ros/gzserver -e ode /home/lbd_song/catkin_ws/src/avoidance/avoidance/sim/worlds/simple_obstacle.world __name:=gazebo __log:=/home/lbd_song/.ros/log/21e44136-484d-11eb-84b3-e0d4e8ff40e7/gazebo-6.log].
log file: /home/lbd_song/.ros/log/21e44136-484d-11eb-84b3-e0d4e8ff40e7/gazebo-6*.log
[ INFO] [1609078169.722340991]: Stereo is NOT SUPPORTED
[ INFO] [1609078169.722418897]: OpenGl version: 4.6 (GLSL 4.6).
[INFO] [1609078169.896500, 0.000000]: Loading model XML from Gazebo Model Database
[INFO] [1609078169.897587, 0.000000]: Waiting for service /gazebo/spawn_sdf_model
[ERROR] [1609078174.312030283]: PR: Unknown parameter to get: MPC_ACC_HOR
[ERROR] [1609078174.313438738]: PR: Unknown parameter to get: MPC_ACC_DOWN_MAX
[ERROR] [1609078174.314867259]: PR: Unknown parameter to get: MPC_ACC_UP_MAX
[ERROR] [1609078174.316075358]: PR: Unknown parameter to get: MPC_XY_CRUISE
[ERROR] [1609078174.317348394]: PR: Unknown parameter to get: MPC_Z_VEL_MAX_DN
[ERROR] [1609078174.318671800]: PR: Unknown parameter to get: MPC_Z_VEL_MAX_UP
[ERROR] [1609078174.319937120]: PR: Unknown parameter to get: CP_DIST
[ERROR] [1609078174.321179755]: PR: Unknown parameter to get: MPC_LAND_SPEED
[ERROR] [1609078174.322409055]: PR: Unknown parameter to get: MPC_JERK_MAX
[ERROR] [1609078174.323600600]: PR: Unknown parameter to get: NAV_ACC_RAD
[ERROR] [1609078174.324799121]: PR: Unknown parameter to get: MPC_YAWRAUTO_MAX

[image]

Now, I cannot figure out what is wrong.
If anyone knows about this problem, or needs more information, please let me know what should I do.

darkasassin44, 2021-01-19:
I couldn't find out the solution, and reinstall the whole ubuntu.
Everything was fine. I think that there was something wrong with my installation.
`[ERROR] [1609078174.324799121]: PR: Unknown parameter to get:` error was RAM problem.

closed 2021-01-19 by darkasassin44

Adhithiyan03, 2022-11-14:
How to run the local planner in airsim