PX4/PX4-Avoidance #619: How to send target point to avoidance in real drone

pirate-lofy, 2021-01-05:
I'm trying to make a drone fly autonomously using avoidance node, Jetson TX2, J120 carrier board and realsense D435i camera..
I'm now in the hardware phase, I used /tools/generate_launchfile.launch.deprecated to generate avoidance file in local_planner node.
it runs without errors now but it reaches a point when nothing happens.
I can arm the vehicle and switch to offbard mode by mavros massages.

```
SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /camera_main/realsense2_camera/accel_fps: 0
 * /camera_main/realsense2_camera/accel_frame_id: camera_main_accel...
 * /camera_main/realsense2_camera/accel_optical_frame_id: camera_main_accel...
 * /camera_main/realsense2_camera/align_depth: False
 * /camera_main/realsense2_camera/aligned_depth_to_color_frame_id: camera_main_align...
 * /camera_main/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_main_align...
 * /camera_main/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_main_align...
 * /camera_main/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_main_align...
 * /camera_main/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_main_align...
 * /camera_main/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_main_align...
 * /camera_main/realsense2_camera/allow_no_texture_points: False
 * /camera_main/realsense2_camera/base_frame_id: camera_main_link
 * /camera_main/realsense2_camera/calib_odom_file: 
 * /camera_main/realsense2_camera/clip_distance: -1.0
 * /camera_main/realsense2_camera/color_fps: 15
 * /camera_main/realsense2_camera/color_frame_id: camera_main_color...
 * /camera_main/realsense2_camera/color_height: 480
 * /camera_main/realsense2_camera/color_optical_frame_id: camera_main_color...
 * /camera_main/realsense2_camera/color_width: 640
 * /camera_main/realsense2_camera/depth_fps: 15
 * /camera_main/realsense2_camera/depth_frame_id: camera_main_depth...
 * /camera_main/realsense2_camera/depth_height: 480
 * /camera_main/realsense2_camera/depth_optical_frame_id: camera_main_depth...
 * /camera_main/realsense2_camera/depth_width: 640
 * /camera_main/realsense2_camera/device_type: 
 * /camera_main/realsense2_camera/enable_accel: False
 * /camera_main/realsense2_camera/enable_color: False
 * /camera_main/realsense2_camera/enable_depth: True
 * /camera_main/realsense2_camera/enable_fisheye1: False
 * /camera_main/realsense2_camera/enable_fisheye2: False
 * /camera_main/realsense2_camera/enable_fisheye: False
 * /camera_main/realsense2_camera/enable_gyro: False
 * /camera_main/realsense2_camera/enable_infra1: True
 * /camera_main/realsense2_camera/enable_infra2: False
 * /camera_main/realsense2_camera/enable_infra: False
 * /camera_main/realsense2_camera/enable_pointcloud: False
 * /camera_main/realsense2_camera/enable_pose: False
 * /camera_main/realsense2_camera/enable_sync: False
 * /camera_main/realsense2_camera/filters: 
 * /camera_main/realsense2_camera/fisheye1_frame_id: camera_main_fishe...
 * /camera_main/realsense2_camera/fisheye1_optical_frame_id: camera_main_fishe...
 * /camera_main/realsense2_camera/fisheye2_frame_id: camera_main_fishe...
 * /camera_main/realsense2_camera/fisheye2_optical_frame_id: camera_main_fishe...
 * /camera_main/realsense2_camera/fisheye_fps: 30
 * /camera_main/realsense2_camera/fisheye_frame_id: camera_main_fishe...
 * /camera_main/realsense2_camera/fisheye_height: 0
 * /camera_main/realsense2_camera/fisheye_optical_frame_id: camera_main_fishe...
 * /camera_main/realsense2_camera/fisheye_width: 0
 * /camera_main/realsense2_camera/gyro_fps: 0
 * /camera_main/realsense2_camera/gyro_frame_id: camera_main_gyro_...
 * /camera_main/realsense2_camera/gyro_optical_frame_id: camera_main_gyro_...
 * /camera_main/realsense2_camera/imu_optical_frame_id: camera_main_imu_o...
 * /camera_main/realsense2_camera/infra1_frame_id: camera_main_infra...
 * /camera_main/realsense2_camera/infra1_optical_frame_id: camera_main_infra...
 * /camera_main/realsense2_camera/infra2_frame_id: camera_main_infra...
 * /camera_main/realsense2_camera/infra2_optical_frame_id: camera_main_infra...
 * /camera_main/realsense2_camera/infra_fps: 15
[... 398 lines cut ...]

[ INFO] [1609876494.481791299]: parameter MPC_ACC_UP_MAX is set from  nan to 4.000000 

[ INFO] [1609876494.501814435]: parameter MPC_JERK_MAX is set from  nan to 8.000000 

[ INFO] [1609876494.501995908]: parameter MPC_JERK_MIN is set from  nan to 8.000000 

[ INFO] [1609876494.585323601]: parameter MPC_TKO_SPEED is set from  nan to 1.500000 
[ WARN] [1609876497.111234795]: PR: request param #416 timeout, retries left 2, and 5 params still missing
[ INFO] [1609876497.731116248]: WP: item #0* F:� C: 22 p: 15 0 0 nan x: 373586884 y: -1219531937 z: 50
[ INFO] [1609876497.734057794]: WP: item #1  F:� C: 16 p: 0 0 0 nan x: 373540421 y: -1219471130 z: 50
[ INFO] [1609876497.734401763]: WP: mission received

```
now I have multiple questions:

1. I have tried to fix
`[ WARN] [1609876485.098644663]: TF Buffer: could not retrieve requested transform from buffer, tf has already been dropped from buffer`
but with no luck, how to fix it and is it really matter or I can skip it?
2. what do the last three lines (WP: -) mean, do they mean it is in mission mode or the node sends target location implicitly?
3. if node sends initial target points and waits for the drone to move and reach them, how it does that?
4. I tried to find local_planner_node to find in which line it sends the target points -if exist- but I couldn't find the file related to local_planner node in recent master branch, is it's deprecated?
5. I tried to publish target points to many topics like /mavros/local_position/pose and /mavros/setpoint_position/local but it didn't work, what is the right topic to do that
6. if it is possible, how to send the target point using mavsdk or even mavros?
thanks.

kiprock, 2021-05-21:
Hello @pirate-lofy I also have this same problem. I can send points in simulation by using R-Vis or just publish directly to /move_base_simple/goal but the planner will not move the real drone. In R-Viz the goal (small yellow ball) does not snow the new goal being set. Again it all works great in simulation, but not in real lifr. I have been also looking at the code but I cannot find out why. Were you able to resolve this?

kiprock, 2021-05-21:
I should say that this is all indoors. Does the module require gps? @mrivi ?

Cristian-wp, 2021-06-17:
I have the same problem using QGC, It did not take...Where you able to solve this?