PX4/PX4-Avoidance #621: safe_landing_planner not working

pirate-lofy, 2021-01-20:
I'm trying to use safe_landing in a real drone, but when trying the simulation files it won't work...
I run `roslaunch safe_landing_planner safe_landing_planner.launch` and I see Rviz world with unmanned vehicle, then start a mission with last waypoint is 'Land' in QGroundController.. drone takes off and follow the mission perfectly, but when lands it lands on obstacles like there is no planner running and this is the result (you can see it landed on a tree and then lost its balance). I tried to land other obstacles with less height and got the same result

[image]

this is the full log
```
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://:14540@loca...
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.0003490659 // 0...
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: local_origin
 * /mavros/local_position/tf/child_frame_id: fcu
 * /mavros/local_position/tf/frame_id: local_origin
 * /mavros/local_position/tf/send: True
 * /mavros/local_position/tf/send_fcu: False
 * /mavros/mission/pull_after_gcs: True
 * /mavros/mocap/use_pose: False
 * /mavros/mocap/use_tf: True
 * /mavros/odometry/fcu/odom_child_id_des: base_link
 * /mavros/odometry/fcu/odom_parent_id_des: map
[... 2671 lines cut ...]
[ INFO] [1611185885.715164648, 38.396000000]: [WGN] Not a land waypoint
[ INFO] [1611185885.715244847, 38.396000000]:  [WGN] goTo nan nan nan - nan nan nan 

[ INFO] [1611185885.715289914, 38.396000000]: [WGN] Landing Radius: xy  nan, z 1.921292 
[ INFO] [1611185885.808836168, 38.488000000]: [WGN] Current position 15.372108 28.772985 7.492333
[ INFO] [1611185885.814409152, 38.492000000]: Input cloud size 3072 
[ INFO] [1611185885.819594923, 38.500000000]: [WGN] Not a land waypoint
[ INFO] [1611185885.909528933, 38.588000000]: [WGN] Current position 15.372941 28.772306 7.487212
[ INFO] [1611185885.910161207, 38.588000000]: Input cloud size 3072 
[ INFO] [1611185885.916126866, 38.596000000]: [WGN] Not a land waypoint
[ INFO] [1611185885.916242781, 38.596000000]:  [WGN] goTo nan nan nan - nan nan nan 

[ INFO] [1611185885.916303259, 38.596000000]: [WGN] Landing Radius: xy  nan, z 1.905005 
INFO  [logger] closed logfile, bytes written: 2268073
[ INFO] [1611185886.010471036, 38.688000000]: [WGN] Current position 15.373319 28.771637 7.481340
[ INFO] [1611185886.013137090, 38.692000000]: Input cloud size 3072 
[ INFO] [1611185886.016131027, 38.692000000]: [WGN] Not a land waypoint
[ INFO] [1611185886.016219255, 38.692000000]:  [WGN] goTo nan nan nan - nan nan nan 

[ INFO] [1611185886.016257035, 38.692000000]: [WGN] Landing Radius: xy  nan, z 1.896451 
[ INFO] [1611185886.112510828, 38.788000000]: [WGN] Current position 15.373356 28.772615 7.477322
[ INFO] [1611185886.115526912, 38.792000000]: Input cloud size 3072 
[ INFO] [1611185886.120828882, 38.796000000]: [WGN] Not a land waypoint
[ INFO] [1611185886.120936844, 38.796000000]:  [WGN] goTo nan nan nan - nan nan nan 
```

AnujAgrawal7, 2021-05-15:
Hey @pirate-lofy were you able to figure out the problem? I am also facing a similar issue.

pirate-lofy, 2021-05-15:
No, i could not

AnujAgrawal7, 2021-05-22:
I figured it out. Unlike the Local and Global Planner the value of `COM_OBS_AVOID` parameter is not set to 1 by default.
Hope this helps!