PX4/PX4-Avoidance #623: Collision Prevention in mission mode

b0b023, 2021-01-21:
Currently "collision prevention" only works in position mode taking in account the OBSTACLE_DISTANCE messages.
Is there a way to make it work in mission mode as well? I think if a simple sensor is present that provide OBSTACLE_DISTANCE messages, it would be great to stop a vehicle even when in mission mode. There is usually no scenario where you don't want the vehicle to crash.

I understand that "obstacle avoidance" via local or global planner is the suggested approach for mission mode.
But often it is better to just stop and don't crash as an alternative to a more complicated and expensive implementation of a planning approach, even if the vehicle can't react further apart from hovering.

Any thoughts on enabling collision prevention mission mode?

baumanta, 2021-01-21:
There once was a draft for this here: https://github.com/PX4/PX4-Autopilot/pull/13260
But it is not a simple thing to do and has quite a few corner cases, especially if you want to use collision prevention as a backup safety measure to the local_planner. The draft as it is, does not perform too bad for SITL, but did not perform well enough in flight tests to be merged. As I did not have the time to look into this anymore, it has gone stale until at some point plans might change again.

poripsa, 2021-02-17:
How is the local_planner working for you in position mode? For me since the values changes position in OBSTACLE_DISTANCE message and if the drone is facing south but a obstacle is in front of it, it thinks its behind it.

@baumanta how do I make sure the values stick to [0] position of the range array in /mavros/obstacle/send OBSTACLE_DISTANCE msg? Right now it moves when yawing the drone to other array positions.

b0b023, 2021-03-11:
@baumanta "The draft as it is, does not perform too bad for SITL, but did not perform well enough in flight tests to be merged." Do you have any insights on what happened - to not repeat those in flight test? :)

baumanta, 2021-03-11:
@b0b023 It was nothing "dangerous", performance just was not nice. The transitions between the avoidance and the collision prevention were too rough and collision prevention interfered in nice avoidance maneuvers due to noise in the depth map sometimes. Stuff like this

b0b023, 2021-03-11:
@baumanta ok, understood - thanks :)