PX4/PX4-Avoidance #624: simulation vehicle

ricardorodrigues09, 2021-02-10:
Hello, I would like to ask a question. I am sorry for creating this thread as an issue but I didn't figure out how to add the question label to it. When I change the vehicle that I want to simulate for both local and global planners, I can only really change it in gazebo, but not in rviz. Would you be kind to tell me if such thing is possible and which file I need to change? I have gone through all the launch files but they all say they are launching an iris drone when in reality that's not the vehicle that appears in rviz. I am a beginner and help would be appreciated.

TonooX, 2021-07-01:
Hi, to change Rviz drone, you have to set the correct model in rviz_world_loader, see below :

https://github.com/PX4/PX4-Avoidance/blob/dafe72aabfdefc2dc409e28f7318717194f1bfa6/avoidance/src/rviz_world_loader.cpp#L134

Hope this helps!

ricardorodrigues09, 2021-08-19:
Thank you very much for your help

closed 2021-08-20 by mrpollo