PX4/PX4-Avoidance #635: Where can i buy it?

yxt916, 2021-03-20:
Intel® Aero Ready to Fly is not selling anymore, so where can i buy the whole thing?

Jaeyoung-Lim, 2021-03-23:
@yxt916 The product is discontinued, and this project doesn't necessarily work with the intel aero

Could you clarify why you want to buy the platform?

yxt916, 2021-03-23:
> @yxt916 The product is discontinued, and this project doesn't necessarily work with the intel aero
> 
> Could you clarify why you want to buy the platform?

Because the localplanner is tested in this drone，if i use the same drone, it will be easy .
Here：
https://github.com/PX4/PX4-Avoidance#companion-computer"
“
Tested models:
local planner: Intel NUC, Jetson TX2, Intel Atom x7-Z8750 (built-in on Intel® Aero RTF drone)
"

Jaeyoung-Lim, 2021-03-23:
@yxt916 I would say a better supported vehicle is http://www.holybro.com/product/px4-vision/

closed 2021-04-02 by yxt916