PX4/PX4-Avoidance #636: TF Buffer error with PX4 firmware master version but work fine with v1.11.3

saengphet, 2021-03-29:
**Describe the bug**
Hello, I installed PX4 avoidance module + D435i. It works fine with PX4 Firmware version v1.11.3. However, after I upgraded Pixhawk to the master version (date 25/03/2021). I got the error message **"TF Buffer: could not retrieve requested transform from buffer, unregistered"**. I try to debug it by plot TF tree. The difference between these two is v1.11.3 has "local_origin" point to "fcu" but version master not. I think this is one of the problems.

**To Reproduce**
Steps to reproduce the behavior:
1. flash firmware master to pixhawk 4
2. connect the fcu to companion computer
3. RUN the local planner node
4. See error, TF Buffer: could not retrieve requested transform from buffer, unregistered"

**Drone (please complete the following information):**
Quadrotor
Pixhawk 4 (also tried the cube orange)
D435i with Jetson NX

TF graph as files attached:

[TF avoidance-v1.11.3.pdf](https://github.com/PX4/PX4-Avoidance/files/6220437/TF.avoidance-v1.11.3.pdf)
[TF avoidance-master.pdf](https://github.com/PX4/PX4-Avoidance/files/6220438/TF.avoidance-master.pdf)

GeneralAdmin, 2021-03-30:
I've also seen a similar issue when doing a fresh install in the simulator yesterday, but, I copied the firmware folder from other computer and compiled it again and it works. Maybe you can use an older version.

saeidfallah1132, 2021-11-27:
@saengphet  
I am using realsense D435 for running   PX4 avoidance ( local planner ), I got the same error **"TF Buffer: could not retrieve requested transform"**.
I am testing on a real drone. 
May I know have you found any solution to fix the issue?

thanks in advance

tiger-k, 2022-04-28:
I got the same error, I think this is TF error ,you can try 
`rosrun tf view_frame`
to see what different betwen real drone and glazbo simulator.
as for me, I use a wrong px4_config.yaml file, 
this is my right config file ,
`cp ..avordance/avordance/reaource/px4_config.yaml ../avoidance/local_plan/reaource/px4_config.yaml`
I hope you can understand whit i say.