PX4/PX4-Avoidance #639: Can't takeoff with ros commands with Avoidance

ataparlar, 2021-04-06:
Hey. 
I am trying to reach a point where i pointed from Rviz with '2D Nav Goal'. I am doing the same things with the Avoidance repo instructions but i couldn't achieve takeoff. Program are not giving any error and saying nothing. Here is a screenshot.
[image]

Same thing happens when i attach a mission from QGC. It is not changing the mode to the MISSION.
I need help with that.

kimzheng1204, 2021-04-08:
Have you solved this problem? I meet the same condition with you.

ataparlar, 2021-04-08:
Not yet. However i realized that no gps signal is publishing when i launch it with `roslaunch local_planner local_planner_stereo.launch`. 
If i start the simulation with `make px4_sitl gazebo` and start the mavros node externally, i get the gps data from **/mavros/global_position/raw/fix** topic.
Also the position of fcu in Rviz is not correct when i try to control with joystick. I think it can't find the position in its own with only IMU.
Maybe the reason of not changing to the MISSION mode in QGC can be this.
Now I'm trying to find a way to get response from GPS.

KimHyungSub, 2021-04-29:
@ataparlar I had the same problem. In my case, I could takeoff and fly the vehicle using QGroundControl.

ataparlar, 2021-05-07:
Thank you for answers. I was using Ubuntu 20. I tried to run the code in Ubuntu 18 and it worked successfully. Now I am able to do everything that i wrote before.

closed 2021-05-07 by ataparlar