PX4/PX4-Avoidance #642: Avoidance always crashes

pirate-lofy, 2021-04-29:
**About the problem**
I used avoidance in simulation for a while and it worked very well, but when i used it with real hardware (jetson tx2/jetson xavier and realsense camera) it only worked for two separated times but most of the time it acts so weird,
I thought it was the pixhawk controller is broken so i used  a new one and got the same results.
I tried HITL with RC and QGC and worked fine.
I managed to make avoidance work in HITL mode then got the same results as the real test.
I only use a script like https://docs.px4.io/master/en/ros/mavros_offboard.html to arm the vehicle and continuously publish the target point

**Problem description**
first drone go up to 4-5 five meters then moves very slow to a position near from the origin then it suddenly moves very fast in a fixed direction (like there is no avoidance system working) and this scenario is repeated every time even if I changed the goal point.

**Expected behavior**
the same as what happens in simulation

**Log Files and Screenshots**
these are three logs from the HITL test
1. https://review.px4.io/plot_app?log=2df68fc9-fa7f-4153-ae14-559bd71c55cb
2. https://review.px4.io/plot_app?log=de0871d5-23af-405f-b7d6-b9f2dd2aec59
3. https://review.px4.io/plot_app?log=4ab1f906-94f8-42d3-9f8b-00fd041b9f7e

**logs**
this is the last part of local_planner logs from HITL test (this logs is usually produced in real test or simulation test)

```

[ INFO] [1619666857.264763582, 47.324000000]: [SP]Tree (59 nodes, 6 path nodes, 40 expanded) calculated in 18.15ms.
[ INFO] [1619666857.340889767, 47.400000000]: [WG] Speed adapted WP: [2.453394 -0.762474 5.806579].
[ INFO] [1619666857.340922993, 47.400000000]: [WG] Final waypoint: [2.467217 -0.729423 5.688729]. nan nan nan 

[ INFO] [1619666857.349254443, 47.408000000]: ===== Set Goal ======: [3.000000, 0.000000, 3.386902].
[ INFO] [1619666857.349290361, 47.408000000]: [OA] Planning started, using 1 cameras
 
[ INFO] [1619666857.364797392, 47.420000000]: [SP]Tree (58 nodes, 7 path nodes, 40 expanded) calculated in 20.08ms.
[ INFO] [1619666857.450467071, 47.500000000]: [WG] Speed adapted WP: [2.448656 -0.757350 5.805224].
[ INFO] [1619666857.450508158, 47.500000000]: [WG] Final waypoint: [2.461490 -0.725807 5.687485]. nan nan nan 

[ INFO] [1619666857.467315297, 47.520000000]: ===== Set Goal ======: [3.000000, 0.000000, 3.386902].
[ INFO] [1619666857.467354974, 47.520000000]: [OA] Planning started, using 1 cameras
 
[ INFO] [1619666857.484949968, 47.540000000]: [SP]Tree (60 nodes, 7 path nodes, 40 expanded) calculated in 11.07ms.
[ INFO] [1619666857.552979906, 47.600000000]: [WG] Speed adapted WP: [2.441467 -0.751906 5.806238].
[ INFO] [1619666857.553044311, 47.600000000]: [WG] Final waypoint: [2.455697 -0.721968 5.686333]. nan nan nan 

[ INFO] [1619666857.577281662, 47.628000000]: ===== Set Goal ======: [3.000000, 0.000000, 3.386902].
[ INFO] [1619666857.577325428, 47.628000000]: [OA] Planning started, using 1 cameras
 
[ INFO] [1619666857.599544569, 47.652000000]: [SP]Tree (62 nodes, 7 path nodes, 40 expanded) calculated in 15.02ms.
[ INFO] [1619666857.645703893, 47.700000000]: [WG] Speed adapted WP: [2.436291 -0.745742 5.804735].
[ INFO] [1619666857.645744745, 47.700000000]: [WG] Final waypoint: [2.449895 -0.717796 5.685137]. nan nan nan 

[ INFO] [1619666857.657224368, 47.712000000]: ===== Set Goal ======: [3.000000, 0.000000, 3.386902].
[ INFO] [1619666857.657271014, 47.712000000]: [OA] Planning started, using 1 cameras
 
[ INFO] [1619666857.668309861, 47.724000000]: [SP]Tree (58 nodes, 7 path nodes, 40 expanded) calculated in 16.88ms.
[ INFO] [1619666857.744814085, 47.800000000]: [WG] Speed adapted WP: [2.431895 -0.738867 5.801560].
[ INFO] [1619666857.744846912, 47.800000000]: [WG] Final waypoint: [2.444235 -0.713267 5.683808]. nan nan nan 

[ INFO] [1619666857.767466186, 47.824000000]: ===== Set Goal ======: [3.000000, 0.000000, 3.386902].
[ INFO] [1619666857.767508161, 47.824000000]: [OA] Planning started, using 1 cameras
 
[ INFO] [1619666857.783216306, 47.840000000]: [SP]Tree (59 nodes, 7 path nodes, 40 expanded) calculated in 10.50ms.
[ INFO] [1619666857.843137490, 47.900000000]: [WG] Speed adapted WP: [2.426433 -0.734032 5.801185].
[ INFO] [1619666857.843185808, 47.900000000]: [WG] Final waypoint: [2.438580 -0.708535 5.682579]. nan nan nan 

[ INFO] [1619666857.877798956, 47.936000000]: ===== Set Goal ======: [3.000000, 0.000000, 3.386902].
[ INFO] [1619666857.877835518, 47.936000000]: [OA] Planning started, using 1 cameras
 
[ INFO] [1619666857.893546547, 47.956000000]: [SP]Tree (58 nodes, 7 path nodes, 40 expanded) calculated in 10.46ms.
[ INFO] [1619666857.938991823, 48.000000000]: [WG] Speed adapted WP: [2.419084 -0.731422 5.803365].
[ INFO] [1619666857.939022804, 48.000000000]: [WG] Final waypoint: [2.432682 -0.703912 5.681536]. nan nan nan 

[... 253 lines cut ...]
[ INFO] [1619666861.862116634, 51.900000000]: [WG] Speed adapted WP: [2.311750 -0.471395 5.782978].
[ INFO] [1619666861.862150147, 51.900000000]: [WG] Final waypoint: [2.313239 -0.463391 5.632195]. nan nan nan 

[ WARN] [1619666876.602265871, 66.500000000]:  Planner abort: missing required data 
 
[ WARN] [1619666876.712058874, 66.600000000]:  Planner abort: missing required data 
 
[ WARN] [1619666876.816496793, 66.700000000]:  Planner abort: missing required data 
 
[ WARN] [1619666876.913689323, 66.800000000]:  Planner abort: missing required data 
 
[ WARN] [1619666877.016486520, 66.900000000]:  Planner abort: missing required data 
 
[ WARN] [1619666877.121670840, 67.000000000]:  Planner abort: missing required data 
 
[ WARN] [1619666877.218722972, 67.100000000]:  Planner abort: missing required data 
 
[ WARN] [1619666877.323198803, 67.200000000]:  Planner abort: missing required data 
 
[ WARN] [1619666877.423749841, 67.300000000]:  Planner abort: missing required data 
 
[ WARN] [1619666877.521893217, 67.400000000]:  Planner abort: missing required data 
 
^C[ WARN] [1619666877.619320282, 67.500000000]:  Planner abort: missing required data 
```

pirate-lofy, 2021-04-30:
after other multiple tests I got this error 
` TF_OLD_DATA ignoring data from the past for frame left_gripper at time 1415804777`
I went to this page http://wiki.ros.org/tf/Errors%20explained and couldn't get a solution 
@mrivi can you help with this?

kiprock, 2021-05-15:
Hey @pirate-lofy  I am getting similar results with the holybro px4 vision (up core + structure core).
Simulation works great but as soon as I take off the drone and put it in offboard mode, it takes off in the same direction every time. I wonder if /local_postion/pose is too erratic in real life for the avoidance algorithm? May I ask how you are sending navigation points to the drone? If I send a 2d nav goal in rviz, it sets Z=0 and in a few tests the drone wanted to fly to the ground.

pirate-lofy, 2021-05-15:
Sorry, I can't figure out whether you talk about simulation or real life test!

kiprock, 2021-05-21:
I left another comment just now on another issue you opened. If I take off the real drone using QGC, set it in offboard it does not crash anymore, it just hovers, but it will not accept goals from the planner. It all works in simulation fine. i tried setting goals using 2d set goal in R-vis as well as publishing goals using /move_base_simple/goal. Have you had any progress?

eldewany, 2021-05-22:
@kiprock The issue is that the local planner is never updated through the topic /mavros/trajectory/desired and thus it always goes to the predefined goal, i tried your method and the goal moves but i haven't tried in real flight so can you share more details about your setup?. @mrivi Can you help with that ? it works fine in sitl but other than that the topic is always empty.

kiprock, 2021-05-26:
Hi @eldewany Thank you for responding. I was able to get it to work. it was my own mistake in the networking configuration as I did not have ROS_IP set on the second computer. I know this is a common problem and I thought I had checked it but no. As for the real flight, my drone was not behaving exactly like the simulation and resulted in several collisions when flying indoors. I would like to know if anyone else has tested Avoidance on real drone indoors. Thank you.

eldewany, 2021-05-26:
@kiprock i don't think indoor navigation is supported for px4 now, at least not without vio or a a position estimator like a Vicon or something. Maybe try it outside in position mode to see if it works fine, then maybe try to run vio along with local planner.