PX4/PX4-Avoidance #643: Error roslaunch local_planner local_planner_depth-camera.launch

lukebing167, 2021-05-20:
Hello, When executing this step, i met this error, drone is at the origin of coordinate but the surrounding trees don't come out. I'd appreciate it if you could help me.

lukebing167, 2021-05-20:
Here is a hint for the terminal.
 roslaunch local_planner local_planner_depth-camera.launch
SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /local_planner_nodelet/goal_x_param: 17
 * /local_planner_nodelet/goal_y_param: 15
 * /local_planner_nodelet/goal_z_param: 3
 * /local_planner_nodelet/pointcloud_topics: ['/camera/depth/p...
 * /local_planner_nodelet/world_name: /home/lukebing/ca...
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://:14540@loca...
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.0003490659 // 0...
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: local_origin
 * /mavros/local_position/tf/child_frame_id: fcu
 * /mavros/local_position/tf/frame_id: local_origin
 * /mavros/local_position/tf/send: True
 * /mavros/local_position/tf/send_fcu: False
 * /mavros/mission/pull_after_gcs: True
 * /mavros/mocap/use_pose: False
 * /mavros/mocap/use_tf: True
 * /mavros/odometry/fcu/odom_child_id_des: base_link
 * /mavros/odometry/fcu/odom_parent_id_des: map
 * /mavros/plugin_blacklist: ['safety_area', '...
 * /mavros/plugin_whitelist: []
 * /mavros/px4flow/frame_id: px4flow
 * /mavros/px4flow/ranger_fov: 0.118682
[... 193 lines cut ...]
[ INFO] [1621553329.783405187]: IMU: High resolution IMU detected!
[INFO] [1621553330.002363, 0.000000]: Loading model XML from Gazebo Model Database
[INFO] [1621553330.004297, 0.000000]: Waiting for service /gazebo/spawn_sdf_model
[ INFO] [1621553330.138380129]: Stereo is NOT SUPPORTED
[ INFO] [1621553330.138443895]: OpenGL device: GeForce GTX 1650/PCIe/SSE2
[ INFO] [1621553330.138464359]: OpenGl version: 4.6 (GLSL 4.6).
[ WARN] [1621553330.268644857]: TM : RTT too high for timesync: 18446742184029.55 ms.
[ WARN] [1621553330.869765480]: TM : RTT too high for timesync: 18446742183429.55 ms.
[ WARN] [1621553331.471457910]: TM : RTT too high for timesync: 18446742182829.55 ms.
[ WARN] [1621553332.074502906]: TM : RTT too high for timesync: 18446742182229.55 ms.
[ WARN] [1621553332.675493505]: TM : RTT too high for timesync: 18446742181629.55 ms.
[ WARN] [1621553333.275924255]: TM : RTT too high for timesync: 18446742181029.55 ms.
[ WARN] [1621553333.877618036]: TM : RTT too high for timesync: 18446742180429.55 ms.
[ WARN] [1621553334.479753019]: TM : RTT too high for timesync: 18446742179829.55 ms.
[ERROR] [1621553334.568188182]: PR: Unknown parameter to get: MPC_ACC_HOR
[ERROR] [1621553334.571345466]: PR: Unknown parameter to get: MPC_ACC_DOWN_MAX
[ERROR] [1621553334.574240841]: PR: Unknown parameter to get: MPC_ACC_UP_MAX
[ERROR] [1621553334.576638103]: PR: Unknown parameter to get: MPC_XY_CRUISE
[ERROR] [1621553334.578917541]: PR: Unknown parameter to get: MPC_Z_VEL_MAX_DN
[ERROR] [1621553334.581383597]: PR: Unknown parameter to get: MPC_Z_VEL_MAX_UP
[ERROR] [1621553334.583815919]: PR: Unknown parameter to get: CP_DIST
[ERROR] [1621553334.585974740]: PR: Unknown parameter to get: MPC_LAND_SPEED
[ERROR] [1621553334.588687967]: PR: Unknown parameter to get: MPC_JERK_MAX
[ERROR] [1621553334.591272964]: PR: Unknown parameter to get: NAV_ACC_RAD
[ERROR] [1621553334.593657794]: PR: Unknown pa

eldewany, 2021-05-22:
I can see that the log spits this "[sitl_0-4] process has died [pid 4619, exit code 255, cmd /home/lukebing/Firmware/build/px4_sitl_default/bin/px4 /home/lukebing/Firmware/build/px4_sitl_default/etc -s etc/init.d-posix/rcS -i 0 __name:=sitl_0 __log:=/home/lukebing/.ros/log/20d92666-b9c3-11eb-8171-a87eea3915bb/sitl_0-4.log]." , maybe try to make a clean rebuild of the whole thing

lukebing167, 2021-05-24:
> I can see that the log spits this
> ""
> [sitl_0-4] process has died [pid 4619, exit code 255, cmd /home/lukebing/Firmware/build/px4_sitl_default/bin/px4 [...]

Thank you very much for your help. I have tried it several times, but it is still the same problem. I wonder if it's the reason why the computer's graphics card result isn't compatible.