PX4/PX4-Avoidance #644: global_planner crashes when OctoMap resolution is below 1m

0l3-git, 2021-05-21:
**Setup:**
I run everything inside a docker container 
- PX4 Firmware version: v1.11.3
- Mavros version: v1.8.0
- Ubuntu: 18.04

**Bug:**
I am using the default global_planner_octomap.launch file to start the global planner. When I tinker with the parameters, more specifically speaking the resolution of the OctoMap and set it below 1m the node crashes a few seconds after takeoff and throws the following error:

`[global_planner_node-1] process has died [pid 32253, exit code -6, cmd /PX4/devel/lib/global_planner/global_planner_node /PX4/src/PX4-Avoidance/global_planner/resource/random_goals __name:=global_planner_node __log:=/home/user/.ros/log/33e3a7ba-ba48-11eb-845b-0242ac110002/global_planner_node-1.log].log file: /home/user/.ros/log/33e3a7ba-ba48-11eb-845b-0242ac110002/global_planner_node-1*.log`

However, no error log is created which makes it hard to investigate the error. As a remedy for this issue I tried to use valgrind with the launch-prefix for the global_planner node. When valgrind is running I am not able to reproduce the error.

Do you have a hint what could cause the error? or a workaround?

DogukanAltay, 2021-07-02:
I am encountering the same error as well. However, the crash is much more random in my case. It can be anytime during my simulations. Trying to investigate the issue for more than 2 months now but still no luck.

Osal22, 2023-04-26:
i am facing the same issue any insight what might be causing the issue