PX4/PX4-Avoidance #645: Failed to run Avoidance "Planner abort: missing required data"

shimajiro4892, 2021-06-01:
After running `roslaunch local_planner local_planner_stereo.launch` , Gazebo and rviz will be launched and the drone and tree models will be displayed.
However, the information from the camera cannot be retrieved and the `Planner abort: missing required data` persistently appears.

I have tried the following steps to solve the problem.
・Check and implement the procedures described in #563  #566 .
・Reinstall ROS and Gazebo.
・Check for the existence of stereo-image-proc.
・Reinstall the OS and rebuild the environment.
I have checked all of the above and have taken action, but cannot resolve the issue.

```
user@user-GE:~$ roslaunch local_planner local_planner_stereo.launch
Checking log directory for disk usage. This may take a while.
... logging to /home/user/.ros/log/ed6e7d00-c28f-11eb-8d55-a85e455aa612/roslaunch-user-GE-6899.log
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
started roslaunch server http://user-GE:39227/
SUMMARY
========
CLEAR PARAMETERS
/mavros/
PARAMETERS
/gazebo/enable_ros_network: True
/local_planner_node/goal_x_param: 15
/local_planner_node/goal_y_param: 15
/local_planner_node/goal_z_param: 4
/local_planner_node/pointcloud_topics: ['/stereo/points2']
/local_planner_node/world_name: /home/user/catkin...
/mavros/cmd/use_comp_id_system_control: False
/mavros/conn/heartbeat_rate: 1.0
/mavros/conn/system_time_rate: 1.0
/mavros/conn/timeout: 10.0
/mavros/conn/timesy```nc_rate: 10.0
/mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
/mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
/mavros/distance_sensor/hrlv_ez4_pub/id: 0
/mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
/mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
/mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
/mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
/mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
/mavros/distance_sensor/laser_1_sub/id: 3
/mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
/mavros/distance_sensor/laser_1_sub/subscriber: True
/mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
/mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
/mavros/distance_sensor/lidarlite_pub/id: 1
/mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
/mavros/distance_sensor/lidarlite_pub/send_tf: True
/mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
/mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
/mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
/mavros/distance_sensor/sonar_1_sub/id: 2
/mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
/mavros/distance_sensor/sonar_1_sub/subscriber: True
/mavros/fcu_protocol: v2.0
/mavros/fcu_url: udp://:14540@loca...
/mavros/gcs_url:
/mavros/global_position/frame_id: fcu
/mavros/global_position/rot_covariance: 99999.0
/mavros/global_position/tf/child_frame_id: fcu_utm
/mavros/global_position/tf/frame_id: local_origin
/mavros/global_position/tf/send: False
/mavros/image/frame_id: px4flow
/mavros/imu/angular_velocity_stdev: 0.0003490659 // 0...
/mavros/imu/frame_id: fcu
/mavros/imu/linear_acceleration_stdev: 0.0003
/mavros/imu/magnetic_stdev: 0.0
/mavros/imu/orientation_stdev: 1.0
/mavros/local_position/frame_id: local_origin
[... 366 lines cut ...]
[ WARN] [1622520915.841985214, 15.460000000]:  Planner abort: missing required data
[ WARN] [1622520915.941575276, 15.560000000]:  Planner abort: missing required data
[ WARN] [1622520916.042613213, 15.660000000]:  Planner abort: missing required data
[ WARN] [1622520916.142463123, 15.760000000]:  Planner abort: missing required data
[ WARN] [1622520916.242591267, 15.860000000]:  Planner abort: missing required data
[ WARN] [1622520916.343367459, 15.960000000]:  Planner abort: missing required data
[ WARN] [1622520916.443197488, 16.060000000]:  Planner abort: missing required data
[ WARN] [1622520916.543645455, 16.160000000]:  Planner abort: missing required data
[ WARN] [1622520916.643927779, 16.260000000]:  Planner abort: missing required data
[ WARN] [1622520916.744515208, 16.360000000]:  Planner abort: missing required data
[ WARN] [1622520916.844604816, 16.460000000]:  Planner abort: missing required data
[ INFO] [1622520916.886333450, 16.500000000]: GF: mission received
[ INFO] [1622520916.887149687, 16.500000000]: RP: mission received
[ INFO] [1622520916.887628489, 16.500000000]: WP: mission received
[ WARN] [1622520916.944704892, 16.560000000]:  Planner abort: missing required data
[ WARN] [1622520917.044912726, 16.660000000]:  Planner abort: missing required data
[ WARN] [1622520917.147688848, 16.760000000]:  Planner abort: missing required data
[ WARN] [1622520917.247865049, 16.860000000]:  Planner abort: missing required data
[ WARN] [1622520917.348329934, 16.960000000]:  Planner abort: missing required data
[ WARN] [1622520917.447938864, 17.060000000]:  Planner abort: missing required data
[ WARN] [1622520917.547475952, 17.160000000]:  Planner abort: missing required data
[ WARN] [1622520917.647540402, 17.260000000]:  Planner abort: missing required data
[ WARN] [1622520917.747071019, 17.360000000]:  Planner abort: missing required data
[ WARN] [1622520917.848081452, 17.460000000]:  Planner abort: missing required data
```

yfaws, 2022-04-08:
I had same issue, 

- I deleted content of ~/.gazebo/models directory
- I copied content of Firmware/sitl_gazebo/models into ~/.gazebo/models
- I cloned https://github.com/osrf/gazebo/tree/gazebo9  and copied  content of models directory in the repo  into ~/.gazebo/models directory, but WITHOUT MERGING - skippping the same directories 

Then ıt solved my problem. 
Maybe, It works for you.

HopeCollector, 2023-11-05:
> I had same issue,
>
> * I deleted content of ~/.gazebo/models directory
> * I copied content of Firmware/sitl_gazebo/models into ~/.gazebo/models
> * I cloned https://github.com/osrf/gazebo/tree/gazebo9 [...]

thank u very much!!!!!!! your advice is all i need!!!!!!!!!

- px4 autopilot v1.13.3
- px4 avodance latest version
- ubuntu 20.04
- gazebo 11
- ros noetic