PX4/PX4-Avoidance #646: AVOIDANCE ON A REAL DRONE

Cristian-wp, 2021-06-16:
Hello, I am working with avoidance (master branch).
In simulation all works, but now I am trying to make it work on a real UAV.
The UAV is a DJI flamewheel 450 with a Holybro pixhawk and an Intel NUC as companion computer mounted on it.
The camera is a IntelRealsense D450i mounted on it with a 3D print gimball (connected with an usbC->usbC cable to the NUC, in this way I can avoid the USB3.0 noise).

As distance sensor I use a Garmin lidar lite.

I have manage to generate the launch file (attached to this thread), and the topic are correctly published and in rviz I can see the pointcloud in /local_poincloud topic and I can see the tree paths generated. I have also compile in release mode, but I can not manage to make it work.

In QGC I set my mission (COM_OBST_AVOID=1) waypoint and start the mission, the problem is that sometime the UAV does not take off, but when it does, It hover and do not do anything....It can not even reach the takeoff height, since is fix at 3m but it fly at 1.5~2.5m and stop.

Can someone help me to solve this problem? 

```
<launch>
    <arg name="ns" default="/"/>
    <arg name="fcu_url" default="/dev/ttyACM0:57600"/>
    <arg name="gcs_url" default="" />   <!-- GCS link is provided by SITL -->
    <arg name="tgt_system" default="1" />
    <arg name="tgt_component" default="1" />

 <!-- Launch static transform publishers -->
  <node pkg="tf" type="static_transform_publisher" name="tf_depth_camera"
          args="0.25 0 0.12 0 0 0 fcu camera_link 10"/>

    <!-- Launch MavROS -->
    <group ns="$(arg ns)">
        <include file="$(find mavros)/launch/node.launch">
            <arg name="pluginlists_yaml" value="$(find mavros)/launch/px4_pluginlists.yaml" />
            <!-- Need to change the config file to get the tf topic and get local position in terms of local origin -->
            <arg name="config_yaml" value="$(find avoidance)/resource/px4_config.yaml" />
            <arg name="fcu_url" value="$(arg fcu_url)" />
            <arg name="gcs_url" value="$(arg gcs_url)" />
            <arg name="tgt_system" value="$(arg tgt_system)" />
            <arg name="tgt_component" value="$(arg tgt_component)" />
        </include>
    </group>

    <!-- Launch cameras -->
   <include file="$(find realsense2_camera)/launch/rs_d435_camera.launch" >
  </include>

    <!-- Launch avoidance -->

  <env name="ROSCONSOLE_CONFIG_FILE" value="$(find local_planner)/resource/custom_rosconsole.conf"/>
  <arg name="pointcloud_topics" default="[/camera/depth/color/points]"/>

    <!-- Launch local planner -->

    <node name="local_planner_node" pkg="local_planner" type="local_planner_node" output="screen" required="true" >
      <param name="goal_x_param" value="0" />
      <param name="goal_y_param" value="0"/>
      <param name="goal_z_param" value="4" />
      <rosparam param="pointcloud_topics" subst_value="True">$(arg pointcloud_topics)</rosparam>
    </node>
    
    <!-- set or toggle rqt parameters -->
    <node name="rqt_param_toggle" pkg="local_planner" type="rqt_param_toggle.sh" />

</launch>
```

julianoes, 2021-06-16:
@Jaeyoung-Lim any idea what is going on here?

@Cristian-wp already said he would upload a log here.

Cristian-wp, 2021-06-16:
My log is to big to be uploaded tomorrow I do another quick test and upload it there

Cristian-wp, 2021-06-17:
@julianoes this are the logs of my tests, I have also a rosbag if is needed.
I have see an old issue so I have set as first point the takeoff at 3m, but the topic /goal_position was always set to 0 0 0 .... maybe there is a comunication problem?

 https://logs.px4.io/plot_app?log=e089088c-a5a0-4f25-81aa-f2beba52035c
https://logs.px4.io/plot_app?log=f527bedc-2ba7-4fbf-90c8-e3aa3611c14e (I havent stop the log after stopping the avoidance)

Cristian-wp, 2021-06-17:
@julianoes @Jaeyoung-Lim Sorry I close it for error

julianoes, 2021-06-17:
So I don't see it go into takeoff mode. How are you commanding it to do something?

Jaeyoung-Lim, 2021-06-17:
@Cristian-wp Are you sure you dont have issues with usb 3.0 interfering with GPS? You might want to try comparing with using a usb 2.0 cable

Cristian-wp, 2021-06-17:
@julianoes I use QGC v.4.1.3, I program the mission starting with a takeoff, after I set some waypoints and at the end a return. If you need I also have a rosbag)

Cristian-wp, 2021-06-17:
@Jaeyoung-Lim Hi! In these test I was using Holybro RTK, because as you say USB3.0 can give noise to GPS.
If I use an USB2 cable the realsense can not strem the point so I can not change the cable...I have already try to use a 2.0 and a C-C usb cable... But On the avoidance official page they use my same camera, how they do?

Cristian-wp, 2021-06-17:
If can help this is the output of my console:

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: False
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: True
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_width: 640
 * /camera/realsense2_camera/initial_reset: True
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
[... 358 lines cut ...]

[ INFO] [1623944798.225607122]: parameter MPC_JERK_MAX is set from  nan to 8.000000 

[ INFO] [1623944798.232649657]: parameter MPC_LAND_SPEED is set from  nan to 0.700000 

[ INFO] [1623944798.236284620]: parameter MPC_TKO_SPEED is set from  nan to 5.000000 

[ INFO] [1623944798.251615851]: parameter MPC_YAWRAUTO_MAX is set from  nan to 45.000000 

[ INFO] [1623944798.252899880]: parameter MPC_Z_VEL_MAX_DN is set from  nan to 1.000000 

[ INFO] [1623944798.252933288]: parameter MPC_Z_VEL_MAX_UP is set from  nan to 3.000000 

[ INFO] [1623944798.253193949]: parameter NAV_ACC_RAD is set from  nan to 2.000000 

[ INFO] [1623944802.965159649]: GF: mission received
[ INFO] [1623944802.965471595]: RP: mission received
[ INFO] [1623944803.070178541]: WP: item #0* F: C: 22 p: 0 0 0 nan x: 451297635 y: 75946219 z: 3
[ INFO] [1623944803.077417271]: WP: item #1  F: C: 16 p: 0 0 0 nan x: 451300864 y: 75945767 z: 3
[ INFO] [1623944803.078506077]: WP: item #2  F: C: 16 p: 0 0 0 nan x: 451298257 y: 75941413 z: 3
[ INFO] [1623944803.148693233]: WP: item #3  F: C: 20 p: 0 0 0 0 x: 0 y: 0 z: 0
[ INFO] [1623944803.148830920]: WP: mission received
[ INFO] [1623944807.917195550]: HP: requesting home position

I copy the output indoor, but I always do my test outside to there is no " HP: requesting home position"

julianoes, 2021-06-17:
In your first log you're always in ALTCTL. Were you able to go into POSCTL? Because if not it probably means you don't have a good position estimate which is also required for mission. 

In the second one it keeps saying that the connection dropped. Do you know why/when that happens? That's also not a good sign. (You can see it on the bottom under "Logged messages".)

And there is the warning "Obstacle avoidance system failed", that also doesn't look good.

[image]

Cristian-wp, 2021-06-18:
Oh my god...My companion computer is connected to the Pixhawk by the standard USB A --> uUSB cable provided...can it be that the problem?

julianoes, 2021-06-18:
I have no idea. It could be various things, that's what you'll have to figure out. One thing to look at is `mavlink status` on the PX4 side to see what is going on.

Cristian-wp, 2021-06-18:
@julianoes I have do it in this moment, (Avoidance active) but I am indoor now, so no GPS signal.

instance #0:
    mavlink chan: #0
    type:        USB CDC
    flow control: OFF
    rates:
      tx: 23.457 kB/s
      txerr: 0.000 kB/s
      tx rate mult: 1.000
      tx rate max: 800000 B/s
      rx: 3.548 kB/s
    FTP enabled: YES, TX enabled: YES
    mode: Config
    MAVLink version: 2
    transport protocol: serial (/dev/ttyACM0 @2000000)

instance #1:
    GCS heartbeat:    846898 us ago
    mavlink chan: #1
    type:        RADIO Link
      rssi:        198
      remote rssi:    193
      txbuf:    98
      noise:    72
      remote noise:    66
      rx errors:    0
      fixed:    0
    flow control: ON
    rates:
      tx: 1.261 kB/s
      txerr: 0.000 kB/s
      tx rate mult: 0.956
      tx rate max: 1200 B/s
      rx: 0.041 kB/s
    FTP enabled: YES, TX enabled: YES
    mode: Normal
    MAVLink version: 2
    transport protocol: serial (/dev/ttyS1 @57600)
    ping statistics:
      last: 143.74 ms
      mean: 212.00 ms
      max: 1150.99 ms
      min: 143.74 ms
      dropped packets: 0

Cristian-wp, 2021-06-18:
I have also perform rostopic echo /diagnostics and this is the output:

rostopic echo /diagnostics
header: 
  seq: 124
  stamp: 
    secs: 1624011759
    nsecs: 621780685
  frame_id: ''
status: 
  - 
    level: 0
    name: "mavros: FCU connection"
    message: "connected"
    hardware_id: "/dev/ttyACM0:57600"
    values: 
      - 
        key: "Received packets:"
        value: "65426"
      - 
        key: "Dropped packets:"
        value: "0"
      - 
        key: "Buffer overruns:"
        value: "0"
      - 
        key: "Parse errors:"
        value: "0"
      - 
        key: "Rx sequence number:"
        value: "160"
      - 
        key: "Tx sequence number:"
        value: "0"
      - 
        key: "Rx total bytes:"
        value: "3778058"
      - 
        key: "Tx total bytes:"
        value: "564008"
      - 
        key: "Rx speed:"
        value: "23320.000000"
      - 
        key: "Tx speed:"
        value: "3609.000000"
  - 
    level: 2
    name: "mavros: GPS"
    message: "No satellites"
    hardware_id: "/dev/ttyACM0:57600"
    values: 
      - 
        key: "Satellites visible"
        value: "0"
      - 
        key: "Fix type"
        value: "0"
      - 
        key: "EPH (m)"
        value: "99.99"
      - 
        key: "EPV (m)"
        value: "99.99"
  - 
    level: 0
    name: "mavros: Heartbeat"
    message: "Normal"
    hardware_id: "/dev/ttyACM0:57600"
    values: 
[... 400 lines cut ...]
      - 
        key: "Errors count #2"
        value: "0"
      - 
        key: "Errors count #3"
        value: "0"
      - 
        key: "Errors count #4"
        value: "0"
  - 
    level: 0
    name: "mavros: Battery"
    message: "Normal"
    hardware_id: "/dev/ttyACM0:57600"
    values: 
      - 
        key: "Voltage"
        value: "18.49"
      - 
        key: "Current"
        value: "0.0"
      - 
        key: "Remaining"
        value: "100.0"
---

Cristian-wp, 2021-06-18:
As suggested in this topic I have also change rosconsol to perform debug on the local planner but I do not get any error

julianoes, 2021-06-18:
Oh right, that all looks fine. Do you see anything else happening when it has this drops?

Cristian-wp, 2021-06-18:
As suggested in a topic I have also change rosconsol to perform debug on the local planner but I do not get any error,
this is the last part:

[ INFO] [1624024340.121066072]: [OA] Planning started, using 1 cameras
 
[ INFO] [1624024340.144858202]: [SP]Tree (114 nodes, 5 path nodes, 40 expanded) calculated in 12.23ms.
[DEBUG] [1624024340.144896978]: [SP] node 0 : [ -0.305969, -1.012261, 0.248882]
[DEBUG] [1624024340.144912561]: [SP] node 1 : [ 1.357684, -0.164588, -0.467854]
[DEBUG] [1624024340.144927612]: [SP] node 2 : [ -0.486489, -0.456676, 0.248882]
[DEBUG] [1624024340.144941076]: [SP] node 3 : [ 0.390553, -0.691679, -1.533131]
[DEBUG] [1624024340.144962292]: [SP] node 4 : [ -0.021662, 0.117338, 0.248882]
[DEBUG] [1624024340.169859801]: [WG] Generate Waypoint, current position: [-0.021669, 0.117347, 0.248709].
[ INFO] [1624024340.169900007]: [WG] Speed adapted WP: [0.000000 0.000000 0.000000].
[DEBUG] [1624024340.169911852]: [WG] Smoothed GoTo location: -0.000158, 0.000793, 0.004097, with dt=0.100085
[ INFO] [1624024340.169927015]: [WG] Final waypoint: [-0.000158 0.000793 0.004097]. nan nan nan 

[DEBUG] [1624024340.169943047]: [WG] Using calculated tree

[DEBUG] [1624024340.169959471]:  [WGN] Update to TRY_PATH state 
 
[ INFO] [1624024340.219626566]: [OA] Planning started, using 1 cameras
 
[ INFO] [1624024340.244878699]: [SP]Tree (114 nodes, 5 path nodes, 40 expanded) calculated in 12.68ms.
[DEBUG] [1624024340.244912705]: [SP] node 0 : [ 0.882278, -0.486908, 0.248701]
[DEBUG] [1624024340.244928881]: [SP] node 1 : [ -1.046926, -0.385802, -0.268937]
[DEBUG] [1624024340.244944073]: [SP] node 2 : [ 0.882278, -0.284697, 0.248701]
[DEBUG] [1624024340.244959192]: [SP] node 3 : [ 0.120781, -0.779219, -1.533312]
[DEBUG] [1624024340.244974669]: [SP] node 4 : [ -0.021259, 0.117583, 0.248701]
[DEBUG] [1624024340.269925460]: [WG] Generate Waypoint, current position: [-0.021121, 0.117651, 0.248617].
[ INFO] [1624024340.270049631]: [WG] Speed adapted WP: [-0.000000 0.000002 0.000004].
[DEBUG] [1624024340.270132959]: [WG] Smoothed GoTo location: -0.000129, 0.000650, 0.003677, with dt=0.100120
[ INFO] [1624024340.270203049]: [WG] Final waypoint: [-0.000129 0.000650 0.003677]. nan nan nan 

[DEBUG] [1624024340.270276469]: [WG] Using calculated tree

[DEBUG] [1624024340.270335589]:  [WGN] Update to TRY_PATH state 
 
[ INFO] [1624024340.325508529]: [OA] Planning started, using 1 cameras
 
[ INFO] [1624024340.357891766]: [SP]Tree (123 nodes, 5 path nodes, 40 expanded) calculated in 14.58ms.
[DEBUG] [1624024340.357926480]: [SP] node 0 : [ -0.305071, -1.011596, 0.248750]
[DEBUG] [1624024340.357941936]: [SP] node 1 : [ 1.358581, -0.163923, -0.467986]
[DEBUG] [1624024340.357955318]: [SP] node 2 : [ -0.485592, -0.456012, 0.248750]
[DEBUG] [1624024340.357982084]: [SP] node 3 : [ 0.391451, -0.691014, -1.533263]
[DEBUG] [1624024340.357996777]: [SP] node 4 : [ -0.020764, 0.118003, 0.248750]
[DEBUG] [1624024340.369859507]: [WG] Generate Waypoint, current position: [-0.020640, 0.118138, 0.248887].
[ INFO] [1624024340.369894854]: [WG] Speed adapted WP: [-0.000028 0.000161 0.000339].
[DEBUG] [1624024340.369903813]: [WG] Smoothed GoTo location: -0.000109, 0.000548, 0.003303, with dt=0.099876
[ INFO] [1624024340.369911061]: [WG] Final waypoint: [-0.000109 0.000548 0.003303]. nan nan nan 

[DEBUG] [1624024340.369919430]: [WG] Using calculated tree

[DEBUG] [1624024340.369927181]:  [WGN] Update to TRY_PATH state 
 
[ INFO] [1624024340.419722871]: [OA] Planning started, using 1 cameras
 
[ INFO] [1624024340.443835165]: [SP]Tree (119 nodes, 5 path nodes, 40 expanded) calculated in 11.96ms.
[DEBUG] [1624024340.443867252]: [SP] node 0 : [ -0.304573, -1.011133, 0.248425]
[DEBUG] [1624024340.443876827]: [SP] node 1 : [ 1.359080, -0.163460, -0.468311]
[DEBUG] [1624024340.443884668]: [SP] node 2 : [ -0.485093, -0.455549, 0.248425]
[DEBUG] [1624024340.443897342]: [SP] node 3 : [ 0.391949, -0.690551, -1.533589]
[DEBUG] [1624024340.443909564]: [SP] node 4 : [ -0.020266, 0.118466, 0.248425]
[DEBUG] [1624024340.469877459]: [WG] Generate Waypoint, current position: [-0.020169, 0.118588, 0.248251].
[ INFO] [1624024340.469979180]: [WG] Speed adapted WP: [0.000000 0.000000 0.000000].
[DEBUG] [1624024340.470045770]: [WG] Smoothed GoTo location: -0.000090, 0.000456, 0.002959, with dt=0.100068
[ INFO] [1624024340.470114413]: [WG] Final waypoint: [-0.000090 0.000456 0.002959]. nan nan nan 

[DEBUG] [1624024340.470179954]: [WG] Using calculated tree

[DEBUG] [1624024340.470259059]:  [WGN] Update to TRY_PATH state 
[... 782 lines cut ...]
[DEBUG] [1624024344.969925748]: [WG] Smoothed GoTo location: -0.000014, 0.000121, 0.001070, with dt=0.100084
[ INFO] [1624024344.969937853]: [WG] Final waypoint: [-0.000014 0.000121 0.001070]. nan nan nan 

[DEBUG] [1624024344.969951279]: [WG] Using calculated tree

[DEBUG] [1624024344.969962829]:  [WGN] Update to TRY_PATH state 
 
[ INFO] [1624024345.022684891]: [OA] Planning started, using 1 cameras
 
[ INFO] [1624024345.047617254]: [SP]Tree (109 nodes, 5 path nodes, 40 expanded) calculated in 12.76ms.
[DEBUG] [1624024345.047654403]: [SP] node 0 : [ -0.291912, -1.015656, 0.247915]
[DEBUG] [1624024345.047671906]: [SP] node 1 : [ 1.371740, -0.167983, -0.468821]
[DEBUG] [1624024345.047686794]: [SP] node 2 : [ -0.472433, -0.460072, 0.247915]
[DEBUG] [1624024345.047705518]: [SP] node 3 : [ 0.404610, -0.695074, -1.534098]
[DEBUG] [1624024345.047723977]: [SP] node 4 : [ -0.007605, 0.113943, 0.247915]
[... cut ...] 07555, 0.113837, 0.247434].
[ INFO] [1624024345.069898246]: [WG] Speed adapted WP: [0.000000 0.000000 0.000000].
[DEBUG] [1624024345.069910553]: [WG] Smoothed GoTo location: -0.000011, 0.000099, 0.000952, with dt=0.099986
[ INFO] [1624024345.069922296]: [WG] Final waypoint: [-0.000011 0.000099 0.000952]. nan nan nan 

[DEBUG] [1624024345.069933246]: [WG] Using calculated tree

[DEBUG] [1624024345.069944562]:  [WGN] Update to TRY_PATH state 

This time I perform the test outdoor, I attach the log

Cristian-wp, 2021-06-18:
@jkflying do you have any ideas?

Jaeyoung-Lim, 2021-06-20:
@Cristian-wp From the launchfile you have attached, the baudrate seems to be set at 57600

This is way below the minimum required data rate for the avoidance to work. Could you try with higher baudrates such as 921600?

Cristian-wp, 2021-06-20:
Hi @Jaeyoung-Lim I have already try to increase it to 921600 and highter values but nothing change....

Jaeyoung-Lim, 2021-06-21:
@Cristian-wp That means you have other issues, the baudrate should be set to 921600 in either case

Cristian-wp, 2021-06-21:
Hi @Jaeyoung-Lim yes, I just try to reduce the fps to avoid USB3.0 and use 2.0 or C. I do this and after I take the new test. (without propeller because I am in the campus and I can not fly now)

Cristian-wp, 2021-06-21:
@Jaeyoung-Lim this is the log https://logs.px4.io/plot_app?log=d0ca338a-b837-4616-8ab2-4383b03884aa

Cristian-wp, 2021-06-21:
Hi @Jaeyoung-Lim , this is the log with 921600 baudrate and the camera connected to an USB 2.0.

https://logs.px4.io/plot_app?log=98fd4e87-a3e7-48fb-8490-b1df5b231ae5

Cristian-wp, 2021-06-21:
This time it seems to work, there is a swith between mission and position because I move the roll stick for error 

https://logs.px4.io/plot_app?log=4043a0b0-2163-47a5-90dd-4bfe7d1af744

Cristian-wp, 2021-06-21:
https://logs.px4.io/plot_app?log=c2e31995-b6ad-4a27-b6e8-6e79fcc9b9bd I have retry immediatly and the error reappear

Cristian-wp, 2021-06-21:
I think I try to change fly controller

Cristian-wp, 2021-06-21:
Ok I have found something really intresting, this is an extract from the log when the avoidance work:
[image]

This is from a case when the avoidance fail:

[image]

As you can see in the correct case there is EKF aligned at 0:00:09, in the wrong you can see at 0:00:13 reset velocity to gps and starting gps fusion, after this messages the obstacle avoidance fails. 
I have just check the other logs and if there are these messages the avoidance does not start...

gajena, 2021-06-24:
But in both logs, data shows similar GPS uncertainty and also similar magnetic interference, not sure why ekf is not getting converged. But maybe try in an open field away from buildings (recalibrate the IMU and Gyro) that might help (not sure though).

Cristian-wp, 2021-06-24:
Hi @gajena I always do my test in an open field :( (same for calibration, no smartphone or other thing near the drone)

Cristian-wp, 2021-12-23:
It was a comunication problem, I have not use the usb-serial adapter.

julianoes, 2021-12-23:
Ok, thanks for reporting back. I'll close this issue then.

closed 2021-12-23 by julianoes