PX4/PX4-Avoidance #649: Drone gets lost during flight with avoidance

eldewany, 2021-06-19:
Hello, i'm trying to fly the drone using local planner and everytime it repeats the same behaviour:

It takes off just fine, tries to go to the required waypoint and like 1 second after it just rotates around itself as if there's obstacles every where. I tried increasing the parameters min_num_points_per_cell_ up to 50 and decreased the frame age to 8 as in here 
[https://github.com/PX4/PX4-Avoidance/issues/309#issuecomment-486588446](url)
there is a parameter mentioned named box_radius but I couldn't find it in dynamic_configure, so I don't know much about it.

I'm using a d435i with a holybro pixhawk 4

this is one of the flight logs 
[https://review.px4.io/plot_app?log=49575a28-d775-47fa-9c8f-6ba477633afe](url)

i will try to upload more logs soon.

eldewany, 2021-06-19:
It's also worth mentioning that i get no errors in any sensor and GPS accuracy is always below 1 meter. Also I fly during the morning and the area is well lit by the sun and checked the camera exposure and everything and the picture it receives is fine.

Cristian-wp, 2021-06-21:
Hi, I having some issue regarding the connection between PX4 and companion computer during mission. Can I ask you to share your launch file and your Pixhawk parameters? (I have your same configuration)

For what concern your problem, have you try to low the value of the pitch? In simulation sometimes get stuck because it try to avoid obstacles only with roll and yaw.

eldewany, 2021-06-21:
@Cristian-wp Hello Cristian, I'm using the default aeroD415 launch file with modification to the dynamic parameters to match the camera mainly :
Max sensor range: 2 m ( From the official d415i , accuracy is high up to 2 m)
min points per cell: 25
Max point age: 20

there is also 2 parameters related to the search algorithm which i still yet to test, but i doubt they will change anything.

Cristian-wp, 2021-06-21:
But you use an intel aero or another companion computer? 
Have you experiment any problem with usb3.0 or EKF with Avoidance?

eldewany, 2021-06-21:
@Cristian-wp  No, I'm using a jetson. The launch file works fine and all the camera parameters are in the realsense ros wrapper file, so there's no need for me to generate a custom launch file. I had some noise in the GPS when the camera was close to it , so I repositioned it somewhere else and it was all good

Cristian-wp, 2021-06-21:
Thanks a lot :) tomorrow I will try!

eldewany, 2021-06-23:
@Cristian-wp Hey, any updates?

Cristian-wp, 2021-07-12:
@eldewany sorry for the late, yes in my case the problem was the communication between flycontroller and companion computer.
Instead of using the USB port I have switch to TELEM2 with a TTL adapter and all works.