PX4/PX4-Avoidance #653: Gazebo Fails to Launch with Local Planner

brunoj6, 2021-08-04:
When running one of the local_planner launch scripts, everything but the gazebo model will run. The corresponding ros topics are also not published leading to empty assets. The rviz world will generate successfully. 

I followed the given documentation to install the workspaces. Occasionally an error message will occur saying that Gazebo is crashed on launch. 

Startup seems to run as normal: 

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /local_planner_node/goal_x_param: 15
 * /local_planner_node/goal_y_param: 15
 * /local_planner_node/goal_z_param: 4
 * /local_planner_node/pointcloud_topics: ['/stereo/points2']
 * /local_planner_node/world_name: /home/joe/catkin_...
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://:14540@loca...
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.0003490659 // 0...
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: local_origin
 * /mavros/local_position/tf/child_frame_id: fcu
 * /mavros/local_position/tf/frame_id: local_origin
 * /mavros/local_position/tf/send: True
 * /mavros/local_position/tf/send_fcu: False
 * /mavros/mission/pull_after_gcs: True
 * /mavros/mocap/use_pose: False
 * /mavros/mocap/use_tf: True
 * /mavros/odometry/fcu/odom_child_id_des: base_link
 * /mavros/odometry/fcu/odom_parent_id_des: map
[... 310 lines cut ...]
[ERROR] [1628103926.006797186, 267.012000000]: PR: Unknown parameter to get: MPC_ACC_HOR
[ERROR] [1628103926.007598151, 267.012000000]: PR: Unknown parameter to get: MPC_ACC_DOWN_MAX
[ERROR] [1628103926.008451277, 267.012000000]: PR: Unknown parameter to get: MPC_ACC_UP_MAX
[ERROR] [1628103926.009264273, 267.016000000]: PR: Unknown parameter to get: MPC_XY_CRUISE
[ERROR] [1628103926.010022907, 267.016000000]: PR: Unknown parameter to get: MPC_Z_VEL_MAX_DN
[ERROR] [1628103926.010711711, 267.016000000]: PR: Unknown parameter to get: MPC_Z_VEL_MAX_UP
[ERROR] [1628103926.011437764, 267.016000000]: PR: Unknown parameter to get: CP_DIST
[ERROR] [1628103926.012165148, 267.016000000]: PR: Unknown parameter to get: MPC_LAND_SPEED
[ERROR] [1628103926.012799840, 267.016000000]: PR: Unknown parameter to get: MPC_JERK_MAX
[ERROR] [1628103926.013417032, 267.020000000]: PR: Unknown parameter to get: NAV_ACC_RAD
[ERROR] [1628103926.014021964, 267.020000000]: PR: Unknown parameter to get: MPC_YAWRAUTO_MAX
[ERROR] [1628103931.015265001, 272.020000000]: PR: Unknown parameter to get: MPC_ACC_HOR
[ERROR] [1628103931.016102967, 272.020000000]: PR: Unknown parameter to get: MPC_ACC_DOWN_MAX
[ERROR] [1628103931.016820771, 272.020000000]: PR: Unknown parameter to get: MPC_ACC_UP_MAX
[ERROR] [1628103931.017498594, 272.020000000]: PR: Unknown parameter to get: MPC_XY_CRUISE
[ERROR] [1628103931.018155137, 272.020000000]: PR: Unknown parameter to get: MPC_Z_VEL_MAX_DN
[ERROR] [1628103931.018841920, 272.020000000]: PR: Unknown parameter to get: MPC_Z_VEL_MAX_UP
[ERROR] [1628103931.019573424, 272.024000000]: PR: Unknown parameter to get: CP_DIST
[ERROR] [1628103931.020174186, 272.024000000]: PR: Unknown parameter to get: MPC_LAND_SPEED
[ERROR] [1628103931.020751547, 272.024000000]: PR: Unknown parameter to get: MPC_JERK_MAX
[ERROR] [1628103931.021330478, 272.024000000]: PR: Unknown parameter to get: NAV_ACC_RAD
[ERROR] [1628103931.021904179, 272.024000000]: PR: Unknown parameter to get: MPC_YAWRAUTO_MAX

Then Rviz will launch without Gazebo:
[image]

aykonsvk, 2021-12-04:
I have the same problems... on ROS Noetic

koby1008, 2022-02-20:
> > I have the same problems... on ROS Noetic
> 
> were you able to solve it?

Can you give me the tutorial how PX4-Avoidance install on ros noetic? Because I follow the PX4-Avoidance github get some trouble

koby1008, 2022-02-20:
> I have the same problems... on ROS Noetic

Can you give me the tutorial how PX4-Avoidance install on ros noetic? Because I follow the PX4-Avoidance github get some trouble

ozangungortuhh, 2022-03-01:
I'm facing the same problem, is there any update on this?

jacobholmes00, 2022-12-21:
> I think Gazebo GUI is set to false by default from the following launch file. By setting it true I was able to get the Gazebo output together with RViz.
> 
> ` <arg name="gui" default="false"/>`
> 
> https://github.com/PX4/PX4-Avoidance/blob/master/avoidance/launch/avoidance_sitl_mavros.launch#L3

Thank you so much!