PX4/PX4-Avoidance #670: I want to contribute

brunopinto900, 2022-03-03:
Hello,

I would like to contribute to this repository since i am very interested in motion planning for drones. I read the issues and seems like the global planner hasn't been tested properly. What is the current state of the global planner and how can i help you? I was thinking of implementing RRT with Octomap and smooth the path with a polynomial spline.

MattGrouchy, 2022-03-06:
@brunopinto900  Hey, I'm a student trying to get the simulator working for the first time. I'm having trouble getting everything working and hit an error when I first build gazebo #669. Would you be able to help me with that?

brunopinto900, 2022-03-09:
@Jaeyoung-Lim Hey i used the pull request from #453 to finish the implementation of RRT and this is the result. There are still things to do before doing a pull request.

[image]

Jaeyoung-Lim, 2022-03-09:
@brunopinto900 Great, what did you add to the pull request?

brunopinto900, 2022-03-14:
@Jaeyoung-Lim i would like some insight on this, if you have the time.

brunopinto900, 2022-03-14:
Hello @Jaeyoung-Lim I will finish the pull request by the end of the week. Then its Open for review and discussion.

Jaeyoung-Lim, 2022-03-16:
@brunopinto900 In principle yes, but I would like to move towards ESDF since it provides better path moving forward.

However, the bigger problem is not the optimal planning framework (we already understand very well what we want), but rather integrating the infrastructure for PX4 which is not straight forward.

We can always improve the planning part after we have sorted the PX4 part out.

brunopinto900, 2022-03-16:
@Jaeyoung-Lim We are on the same boat then, because i want to move torwards ESDF. So the goal now is to finish the pull request. After that we shall discuss the further details.

brunopinto900, 2022-03-18:
@Jaeyoung-Lim I did the PR. In the meanwhile, what things can i do? Think about the ESDF?

Do you want to use voxblox or use code from FastPlanner https://github.com/HKUST-Aerial-Robotics/Fast-Planner ?

Thank you.

Jaeyoung-Lim, 2022-03-18:
@brunopinto900 I think the next step would be trying to make the planner work in mission mode with OMPL before we move on to a different map representation or planner

brunopinto900, 2022-03-18:
I am not familiar with mission mode. What modifications are necessary to make it work in mission mode?

Jaeyoung-Lim, 2022-03-18:
@brunopinto900 We need to get the planner to work with this interface: https://docs.px4.io/master/en/computer_vision/path_planning_interface.html

brunopinto900, 2022-03-18:
Ah i see. Unfortunely i don't have a vehicle (only in the summer).

brunopinto900, 2022-03-18:
> @brunopinto900 We need to get the planner to work with this interface: https://docs.px4.io/master/en/computer_vision/path_planning_interface.html , which is made so that such planner can work with mission mode
> 

@Jaeyoung-Lim Can you point some where this is done on the code? Although i understand the concept of mission mode i would like some pointers and see where it is in the code.

brunopinto900, 2022-03-21:
@Jaeyoung-Lim Regarding the implementation of the state machine for the RRT planner, which is specially important in mission mode. Why did the PR #506 didn't went through? How reasonable is to port this PR into the RRT planner?

Thank you.