PX4/PX4-Avoidance #671: avoidance target already exist in PX4-Avoidance

iamrajee, 2022-03-04:
I'm getting the below error while building the PX4-Avoidance package on my ubuntu.

[image]

```
-- Building avoidance with Gazebo Simulation
CMake Error at PX4-Avoidance/avoidance/CMakeLists.txt:63 (add_library):
  add_library cannot create target "avoidance" because another target with
  the same name already exists.  The existing target is a static library
  created in source directory
  "/home/rajendra/drone_ws/src/PX4-Autopilot/src/lib/avoidance".  See
  documentation for policy CMP0002 for more details.

CMake Error at PX4-Avoidance/avoidance/CMakeLists.txt:70 (target_link_libraries):
  The keyword signature for target_link_libraries has already been used with
  the target "avoidance".  All uses of target_link_libraries with a target
  must be either all-keyword or all-plain.

  The uses of the keyword signature are here:

   * PX4-Autopilot/cmake/px4_add_library.cmake:48 (target_link_libraries)
   * PX4-Autopilot/src/lib/avoidance/CMakeLists.txt:37 (target_link_libraries)
  ```

My configuration:

ros_ws/src
        |---------------PX4-Autopilot
        |---------------PX4-SITL_gazebo
        |---------------[PX4-Avoidance](https://github.com/PX4/PX4-Avoidance)

Then, I tried to build using catkin_make in my ros_ws. (Note: I can run SITL with such configuration.)

Thinks I tried:
tried to remove PX4-Autopilot/src/lib/avoidance then tried to build again. But apparently, it was required in CMakeList of PX4-Autopilot, thus there should be an easier way.

Thank you, your help in this would be of great help.

Jaeyoung-Lim, 2022-03-04:
@iamrajee There seems to be several things wrong with your workspace

- The PX4 Autopilot should not be inside your catkin workspace
- sitl_gazebo is a submodule of PX4 Autopilot, therefore you don't need to have an additional clone of the repository

iamrajee, 2022-03-29:
> @iamrajee There seems to be several things wrong with your workspace
> 
> * The PX4 Autopilot should not be inside your catkin workspace
> * sitl_gazebo is a submodule of PX4 Autopilot, therefore you don't need to have an additional clone of the repository

Thank you, I tried to config the way you suggested and the issue got resolved.

closed 2022-03-29 by iamrajee