PX4/PX4-Avoidance #673: how to make local_planner preffer to go above or below obstacle rather than around it?

DoneMan, 2022-03-20:
HI all.

i've been using local_planner with a lot of success, it is truly some amazing piece of code. i wanted to ask: is it possible to configure somehow to tell the local_planner to prefer going above or below obstacles rather then around them? 
if i put a wall of obstacles in front of the drone, it will prefer to go around them even if the wall is 400 meters wide, if it would just go above or below it will be much better off.

thanks in advanced

DoneMan, 2022-03-23:
@dagar @dayjaby @bkueng

jhiggason, 2023-01-04:
Did anyone here get this figured out by chance?  If so- I'd love to know how you did it. =)  Thank you!

aravi-2001, 2023-10-11:
reduce the pitch cost param and increase yaw cost