PX4/PX4-Avoidance #676: Global planner with lidar benewake (jetson Xavier)

dmeirap, 2022-03-29:
Hiii!!! I'm trying to connect my lidar benewake with the global planner node in my jetson Xavier without success. Do you know anything about the error i'm facing up? 

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /ce30_cloud/IP: 192.168.1.80
 * /ce30_cloud/frame_id: map
 * /global_planner_node/camera_frame_id: /ce30_cloud
 * /global_planner_node/frame_id: local_origin
 * /global_planner_node/pointcloud_topics: ['/ce30c_output']
 * /global_planner_node/robot_radius: 0.5
 * /global_planner_node/start_pos_x: 0
 * /global_planner_node/start_pos_y: 0
 * /global_planner_node/start_pos_z: 3.0
 * /mavros/camera/frame_id: base_link
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/horizontal_fov_ratio: 1.0
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/distance_sensor/sonar_1_sub/vertical_fov_ratio: 1.0
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: /dev/ttyUSB0:921600
 * /mavros/gcs_url: 
 * /mavros/global_position/child_frame_id: base_link
 * /mavros/global_position/frame_id: map
 * /mavros/global_position/gps_uere: 1.0
[... 275 lines cut ...]
[ERROR] [1648542037.367895766]: PR: Unknown parameter to get: MPC_XY_CRUISE
[ERROR] [1648542037.409432290]: PR: Unknown parameter to get: MPC_Z_VEL_MAX_DN
[ERROR] [1648542037.430657980]: PR: Unknown parameter to get: MPC_Z_VEL_MAX_UP
[ERROR] [1648542037.457868139]: PR: Unknown parameter to get: CP_DIST
[ERROR] [1648542037.472672876]: PR: Unknown parameter to get: MPC_LAND_SPEED
[ERROR] [1648542037.493074111]: PR: Unknown parameter to get: MPC_JERK_MAX
[ERROR] [1648542037.513540179]: PR: Unknown parameter to get: NAV_ACC_RAD
[ERROR] [1648542037.525071832]: PR: Unknown parameter to get: MPC_YAWRAUTO_MAX
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
[global_planner_node-5] process has died [pid 9863, exit code -6, cmd /home/nvidia/catkin_ws/devel/lib/global_planner/global_planner_node __name:=global_planner_node __log:=/home/nvidia/.ros/log/13cfb3f6-af39-11ec-9de7-565b1a51901d/global_planner_node-5.log].
log file: /home/nvidia/.ros/log/13cfb3f6-af39-11ec-9de7-565b1a51901d/global_planner_node-5*.log
^C[rviz-7] killing on exit
[octomap_server-6] killing on exit
[ce30_cloud-4] killing on exit
[mavros-3] killing on exit
ERROR: Failed to receive distance data!
[lidar_benewake-2] killing on exit
Measurement is stopped!
LiDAR is disconnected!
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

dmeirap, 2022-03-29:
--- real_planner.launch ---

<?xml version="1.0"?>
<launch>

    <arg name="pointcloud_topics" default="[/ce30c_output]"/>
    <arg name="camera_frame_id" default="ce30_cloud"/>
    <arg name="start_pos_x" default="0" />
    <arg name="start_pos_y" default="0" />
    <arg name="start_pos_z" default="3.0" />

    
    
    <node pkg="tf" type="static_transform_publisher" name="lidar_benewake" args="0 0 0 0 0 0 fcu $(arg camera_frame_id) 100" />

    
    <include file="real_px4.launch" />

    
    <include file="real_octomap.launch" >
        <arg name="pointcloud_topics" value="$(arg pointcloud_topics)" />
        <arg name="camera_frame_id" value="$(arg camera_frame_id)" />
        <arg name="start_pos_x" value="$(arg start_pos_x)" />
        <arg name="start_pos_y" value="$(arg start_pos_y)" />
        <arg name="start_pos_z" value="$(arg start_pos_z)" />
    </include>>

    <node pkg="rviz" type="rviz" output="screen" name="rviz" respawn="true"
          args="-d $(find global_planner)/resource/global_planner.rviz"/>

</launch>

dmeirap, 2022-03-29:
--- real_px4.launch ---

<?xml version="1.0"?>
<launch>
	<arg name="fcu_url" default="/dev/ttyUSB0:921600" /> 
	<arg name="gcs_url" default="" />
	<arg name="tgt_system" default="1" />
	<arg name="tgt_component" default="1" />
	<arg name="log_output" default="screen" />
	<arg name="fcu_protocol" default="v2.0" />
	<arg name="respawn_mavros" default="false" />

	<include file="$(find mavros)/launch/node.launch">

		<arg name="pluginlists_yaml" value="$(find mavros)/launch/px4_pluginlists.yaml" />
		<arg name="config_yaml" value="$(find mavros)/launch/px4_config.yaml" />

		<arg name="fcu_url" value="$(arg fcu_url)" />
		<arg name="gcs_url" value="$(arg gcs_url)" />
		<arg name="tgt_system" value="$(arg tgt_system)" />
		<arg name="tgt_component" value="$(arg tgt_component)" />
		<arg name="log_output" value="$(arg log_output)" />
		<arg name="fcu_protocol" value="$(arg fcu_protocol)" />
		<arg name="respawn_mavros" default="$(arg respawn_mavros)" />

	</include>
	
	<param name="/mavros/conn/timesync_rate" value="0.0" /> 

	
	<node pkg="ce30c_driver" type="ce30_cloud" name="ce30_cloud" output="screen" >
		<param name="IP" type="string" value="192.168.1.80" />
		<param name="frame_id" type="string" value="map" />
	</node>
</launch>

dmeirap, 2022-03-29:
--- real_octomap.launch ---

<?xml version="1.0"?>
<launch>
    <arg name="pointcloud_topics" default="[/ce30c_output]]"/>
    <arg name="camera_frame_id" default="ce30_cloud"/>
    <arg name="start_pos_x" default="0.0" />
    <arg name="start_pos_y" default="0.0" />
    <arg name="start_pos_z" default="3.0" />
    <arg name="frame_id"    default="local_origin" />

    
    <node name="global_planner_node" pkg="global_planner" type="global_planner_node">

        <param name="frame_id" type="string" value="$(arg frame_id)" />
        <param name="camera_frame_id" type="string" value="/$(arg camera_frame_id)" />
        <param name="start_pos_x" value="$(arg start_pos_x)" />
        <param name="start_pos_y" value="$(arg start_pos_y)" />
        <param name="start_pos_z" value="$(arg start_pos_z)" />
        <rosparam param="pointcloud_topics" subst_value="True">$(arg pointcloud_topics)</rosparam>
        <param name="robot_radius" value="0.5" /> 

    </node>

    
    <node pkg="octomap_server" type="octomap_server_node" name="octomap_server">
        <param name="resolution" value="1.0" />
        <param name="frame_id" type="string" value="$(arg frame_id)" />
        <param name="sensor_model/max_range" value="9.0" />
        <param name="sensor_model/min" value="0.05" />
        <param name="sensor_model/max" value="0.99" />
        <param name="sensor_model/hit" value="0.9" />
        <param name="sensor_model/miss" value="0.45" />
        <param name="color/r" value="0.1" />
        <param name="color/g" value="0.1" />
        <param name="color/b" value="0.1" />
        <param name="color/a" value="1.0" />
        <param name="occupancy_min_z" value="0.5" />
        <param name="height_map" value="true" /> 
        <param name="publish_free_space" value="false" />
        
            <remap from="cloud_in" to="/ce30c_output"/>
            
    </node>
</launch>

anunarayan13, 2023-07-10:
Have you come across any solutions?