PX4/PX4-Avoidance #677: local_origin not successfully connected with fcu

yubo105139, 2022-04-29:
**Describe the bug**
Hello, I am using this program for differential trolley operation. I am using realsense d455, px4 solid v1.11.3, but still there is local_origin not successfully connected with fcu. What is the situation please?

**Expected behavior**
I want my little car to be able to move

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://hewen-desktop:36641/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /local_planner_node/goal_x_param: 0
 * /local_planner_node/goal_y_param: 0
 * /local_planner_node/goal_z_param: 4
 * /local_planner_node/pointcloud_topics: ['/camera/depth/c...
 * /mavros/camera/frame_id: base_link
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/horizontal_fov_ratio: 1.0
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/distance_sensor/sonar_1_sub/vertical_fov_ratio: 1.0
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_mocap: True
[... 157 lines cut ...]
[ INFO] [1651222299.308464861]: Plugin esc_status loaded
[ INFO] [1651222299.347682537]: Plugin esc_status initialized
[ INFO] [1651222299.348019399]: Plugin esc_telemetry loaded
[ INFO] [1651222299.366732153]: Plugin esc_telemetry initialized
[ INFO] [1651222299.367243122]: Plugin fake_gps loaded
[DEBUG] [1651222299.927097196]: Trying to publish message of type [dynamic_reconfigure/ConfigDescription/757ce9d44ba8ddd801bb30bc456f946f] on a publisher with type [dynamic_reconfigure/ConfigDescription/757ce9d44ba8ddd801bb30bc456f946f]
[ INFO] [1651222300.181567515]: Plugin fake_gps initialized
[ INFO] [1651222300.181686730]: Plugin ftp blacklisted
[ INFO] [1651222300.182037185]: Plugin geofence loaded
[ INFO] [1651222300.299885770]: Plugin geofence initialized
[ INFO] [1651222300.300436478]: Plugin global_position loaded
[ INFO] [1651222300.447543038]: Stereo is NOT SUPPORTED
source_frame_data camera_depth_optical_frame, target_frame /fcu
[ INFO] [1651222300.447933492]: OpenGL device: NVIDIA Tegra X1 (nvgpu)/integrated
[ INFO] [1651222300.448051300]: OpenGl version: 4.6 (GLSL 4.6).
[DEBUG] [1651222300.819215641]: Trying to publish message of type [dynamic_reconfigure/Config/958f16a05573709014982821e6822580] on a publisher with type [dynamic_reconfigure/Config/958f16a05573709014982821e6822580]
[ INFO] [1651222300.819701664]: ===== Set Goal ======: [0.000000, 0.000000, 4.000000].
[DEBUG] [1651222300.819867542]: [OA] Dynamic reconfigure call 
[ERROR] [1651222300.878172979]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1651222300.919445382]: TF Buffer: could not retrieve requested transform from buffer, unregistered

**Additional context**
run jetson nano
[frames.pdf](https://github.com/PX4/PX4-Avoidance/files/8589563/frames.pdf)
[Uploading frames.pdf…]()

wlgnsthd, 2022-09-04:
I solved this problem by adding local_origin tf arg in local planner launcher.
[image]
[image]

And I have a question about my frame tree relevance.
Could you let me know about your result frame tree after solving the problem?

Badrivishal, 2022-12-23:
This would not help as the transform from local origin to the fcu is not supposed to be static. It will be published by mavros. To check it, check whether the local_posistion/pose topic is being published. If it is, check the px4_config.yaml file for the transform.

>