PX4/PX4-Avoidance #679: Hardware implementation Generate launch file is deprecated

Samyak-ja-in, 2022-07-16:
I was trying to implement local planner on hardware but I am unable to generate launch file as the file 
generate_launchfile.sh
is deprecated
Thanks!

Jaeyoung-Lim, 2022-07-16:
@Samyak-ja-in the lauchfile generation script is not necessary to run the package on hardware

Samyak-ja-in, 2022-07-17:
So how to run it on hardware I am stuck can you share me the launch file to launch

Jaeyoung-Lim, 2022-07-17:
@Samyak-ja-in You need to write a launchfile that fits your hardware. The launchfile generation does not solve this problem for you

Samyak-ja-in, 2022-07-17:
I am also unable to find px4_config.yaml at the given path
\$(find local_planner)/resource/px4_config.yaml"

Jaeyoung-Lim, 2022-07-18:
@Samyak-ja-in which file is using that path?

Samyak-ja-in, 2022-07-19:
@Jaeyoung-Lim generate_launchfile_deprecated.sh
I am trying to generate launch file using that deprecated file.
I solved it by copying px4_config.yaml from global_planner

Samyak-ja-in, 2022-07-19:
But now getting another error after launching avoidance.launch file

Jaeyoung-Lim, 2022-07-19:
@Samyak-ja-in The launchfile generation script is deprecated therefore is not expected to work. The file you mentioned had been moved previously, but the launchfile generation script had not been updated since it has been deprecated.

If you have other issues please post a new issue

closed 2022-07-19 by Jaeyoung-Lim